dusty-nv/jetson-containers #345: Jetson Nano Ros2 build failed with ros2_control package: c++ 8 charconv

mischa-ai, 2023-12-09:
Jetson Nano with

Jetpack 4.6.4
L4T 32.7.4
Docker 24.0.2

I tried to build the `ros2_control` package from source in the `dustynv/ros:iron-ros-base-l4t-r32.7.1` container and also to build ros2 completely with the package with the same outcome.

To build completely, I used basically this [Dockerfile](https://github.com/dusty-nv/jetson-containers/blob/master/packages/ros/Dockerfile.ros2) and this [script](https://github.com/dusty-nv/jetson-containers/blob/master/packages/ros/ros2_build.sh) and just added `ros2_control` and `ros2_controllers` to the `rosinstall_generator` step.

In both ways this error occured:

```
#0 8827.7 Finished <<< rclcpp_components [4min 42s]
#0 8827.7 Starting >>> rosbag2_test_common
#0 8844.2 Finished <<< rosbag2_test_common [16.5s]
#0 8844.2 Starting >>> rosbag2_storage
#0 8901.4 --- stderr: cv_bridge
#0 8901.4 CMake Warning (dev) at /usr/lib/cmake/opencv4/OpenCVConfig.cmake:86 (find_package):
#0 8901.4   Policy CMP0146 is not set: The FindCUDA module is removed.  Run "cmake
#0 8901.4   --help-policy CMP0146" for policy details.  Use the cmake_policy command to
#0 8901.4   set the policy and suppress this warning.
#0 8901.4 
#0 8901.4 Call Stack (most recent call first):
#0 8901.4   /usr/lib/cmake/opencv4/OpenCVConfig.cmake:108 (find_host_package)
#0 8901.4   CMakeLists.txt:40 (find_package)
#0 8901.4 This warning is for project developers.  Use -Wno-dev to suppress it.
#0 8901.4 
#0 8901.4 ---
#0 8901.4 [Processing: cv_bridge, rclcpp_action, rclcpp_lifecycle, rosbag2_storage]
#0 8901.4 Finished <<< cv_bridge [5min 55s]
#0 8901.4 Starting >>> camera_calibration_parsers
#0 8949.0 [Processing: camera_calibration_parsers, rclcpp_action, rclcpp_lifecycle, rosbag2_storage]
#0 8949.0 Finished <<< rclcpp_action [6min 43s]
#0 8949.0 Starting >>> realtime_tools
#0 9036.5 [Processing: camera_calibration_parsers, rclcpp_lifecycle, realtime_tools, rosbag2_storage]
#0 9036.5 [Processing: camera_calibration_parsers, rclcpp_lifecycle, realtime_tools, rosbag2_storage]
#0 9036.5 Finished <<< camera_calibration_parsers [2min 15s]
#0 9036.5 Starting >>> ros2component
#0 9046.6 --- stderr: ros2component
#0 9046.6 /usr/local/lib/python3.6/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
#0 9046.6   setuptools.SetuptoolsDeprecationWarning,
#0 9046.6 ---
#0 9046.6 Finished <<< ros2component [10.1s]
#0 9046.6 Starting >>> vision_opencv
#0 9053.2 Finished <<< rosbag2_storage [3min 29s]
#0 9053.2 Starting >>> rosbag2_storage_mcap
#0 9056.7 Finished <<< vision_opencv [10.1s]
#0 9056.7 Starting >>> rosbag2_storage_sqlite3
#0 9079.1 Finished <<< rclcpp_lifecycle [8min 53s]
#0 9079.1 Starting >>> message_filters
#0 9188.7 [Processing: message_filters, realtime_tools, rosbag2_storage_mcap, rosbag2_storage_sqlite3]
#0 9188.7 [Processing: message_filters, realtime_tools, rosbag2_storage_mcap, rosbag2_storage_sqlite3]
#0 9188.7 [Processing: message_filters, realtime_tools, rosbag2_storage_mcap, rosbag2_storage_sqlite3]
#0 9188.7 Finished <<< rosbag2_storage_mcap [2min 16s]
#0 9188.8 Starting >>> hardware_interface
#0 9191.4 --- stderr: realtime_tools
#0 9191.4 /opt/ros/iron/src/realtime_tools/test/realtime_server_goal_handle_tests.cpp: In function ‘std::shared_ptr<rclcpp_action::ClientGoalHandle<test_msgs::action::Fibonacci> > send_goal(rclcpp::Node::SharedPtr, std::shared_ptr<rclcpp_action::Client<test_msgs::action::Fibonacci> >, const string&, ActionClientCallbacks&)’:
#0 9191.4 /opt/ros/iron/src/realtime_tools/test/realtime_server_goal_handle_tests.cpp:127:23: warning: unused parameter ‘server_name’ [-Wunused-parameter]
#0 9191.4    const std::string & server_name, ActionClientCallbacks & client_callbacks)
#0 9191.4    ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
#0 9191.4 ---
#0 9191.4 Finished <<< realtime_tools [4min 2s]
#0 9191.5 Starting >>> camera_info_manager
#0 9221.2 --- stderr: hardware_interface
#0 9221.2 /opt/ros/iron/src/ros2_control/hardware_interface/src/component_parser.cpp: In function ‘double hardware_interface::detail::get_parameter_value_or(const tinyxml2::XMLElement*, const char*, double)’:
#0 9221.2 /opt/ros/iron/src/ros2_control/hardware_interface/src/component_parser.cpp:141:83: error: no matching function for call to ‘from_chars(const char* const&, const char*, double&)’
#0 9221.2            std::from_chars(tag_text, tag_text + std::strlen(tag_text), result_value);
#0 9221.2                                                                                    ^
#0 9221.2 In file included from /opt/ros/iron/src/ros2_control/hardware_interface/src/component_parser.cpp:16:
#0 9221.2 /usr/include/c++/8/charconv:593:5: note: candidate: ‘template<class _Tp> std::__detail::__integer_from_chars_result_type<_Tp> std::from_chars(const char*, const char*, _Tp&, int)’
#0 9221.2      from_chars(const char* __first, const char* __last, _Tp& __value,
#0 9221.2      ^~~~~~~~~~
#0 9221.2 /usr/include/c++/8/charconv:593:5: note:   template argument deduction/substitution failed:
#0 9221.2 make[2]: *** [CMakeFiles/hardware_interface.dir/src/component_parser.cpp.o] Error 1
#0 9221.2 make[1]: *** [CMakeFiles/hardware_interface.dir/all] Error 2
#0 9221.2 make: *** [all] Error 2
#0 9221.2 ---
#0 9221.2 Failed   <<< hardware_interface [32.5s, exited with code 2]
#0 9225.9 Aborted  <<< camera_info_manager [34.4s]
#0 9242.8 Aborted  <<< rosbag2_storage_sqlite3 [3min 6s]
#0 9382.9 Aborted  <<< message_filters [5min 4s]
#0 9384.2 
#0 9384.2 Summary: 215 packages finished [2h 19min 28s]
#0 9384.2   1 package failed: hardware_interface
#0 9384.2   3 packages aborted: camera_info_manager message_filters rosbag2_storage_sqlite3
#0 9384.2   96 packages had stderr output: action_msgs actionlib_msgs ament_clang_format ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pep257 ament_pycodestyle ament_uncrustify ament_xmllint builtin_interfaces composition_interfaces control_msgs controller_manager_msgs cv_bridge demo_nodes_py diagnostic_msgs domain_coordinator example_interfaces geometry_msgs google_benchmark_vendor hardware_interface iceoryx_posh image_geometry launch launch_ros launch_testing launch_testing_ament_cmake launch_testing_ros launch_xml launch_yaml libstatistics_collector lifecycle_msgs message_filters mimick_vendor nav_msgs osrf_pycommon osrf_testing_tools_cpp rcl rcl_interfaces rclcpp rcutils realtime_tools rmw_connextdds_common rmw_dds_common ros2action ros2cli ros2cli_test_interfaces ros2component ros2doctor ros2interface ros2launch ros2multicast ros2node ros2param ros2pkg ros2run ros2service ros2test ros2topic rosbag2_interfaces rosbag2_test_msgdefs rosgraph_msgs rosidl_cli rosidl_generator_py rosidl_pycommon rosidl_runtime_py rpyutils rti_connext_dds_cmake_module sensor_msgs sensor_msgs_py service_msgs shape_msgs sros2 statistics_msgs std_msgs std_srvs stereo_msgs test_msgs tf2_msgs tf2_ros_py tf2_tools tracetools_image_pipeline trajectory_msgs type_description_interfaces uncrustify_vendor unique_identifier_msgs urdfdom_headers vision_msgs visualization_msgs
#0 9384.2   42 packages not processed
------
Dockerfile:31
--------------------
  29 |     # build ROS from source
  30 |     COPY ros2_build.sh ros2_build.sh
  31 | >>> RUN bash ./ros2_build.sh
  32 |     
  33 |     # Set the default DDS middleware to cyclonedds
--------------------
ERROR: failed to solve: process "/bin/bash -c bash ./ros2_build.sh" did not complete successfully: exit code: 2
```

I tried to find out what it is, and it seems to be a problem with C++17 ... but I have no idea how to proceed further to resolve that problem.

The script seems to install gcc8 but: 

```
echo "Compiler version: $(gcc --version)"
echo "CMake version: $(cmake --version)"
```

```
Compiler version: gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
CMake version: cmake version 3.27.0
```

Any ideas @dusty-nv ?

dusty-nv, 2023-12-11:
> The script seems to install gcc8 but:
> 
> ```
> echo "Compiler version: $(gcc --version)"
> echo "CMake version: $(cmake --version)"
> ```
> 
> ```
> Compiler version: gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
> CMake version: cmake version 3.27.0
> ```

Hi @mischa-ai, do you see this print line get triggered in the build?

https://github.com/dusty-nv/jetson-containers/blob/93739890ec4f575c113e0c8cfad76a6e4e1a51d3/packages/ros/ros2_build.sh#L118

On 18.04, it will install gcc-8 for Humble/Iron, and sets the CC/CXX environment vars that CMake picks up and compiles with (even though `gcc` still points to the original).  If that wasn't working, you may have run into build errors even earlier that were related to C++17 support.

Since your issue seems to emanate from the `ros2_control` package you are adding, my guess is that package requires an even newer compiler version.  gcc-8 is the latest that's in the 18.04 apt repo, however maybe you could find a newer one via a PPA like here:  https://askubuntu.com/a/1149383

It's also probably worth mentioning that 18.04 is a bit long in the tooth for full ROS2 Iron, which is not formally supported on 18.04:  https://www.ros.org/reps/rep-2000.html#iron-irwini-may-2023-november-2024   so if you require this package, you may need to either patch it manually (if other methods of upgrading the toolchain don't work for you), or roll back to an older version of ROS.

elgarbe, 2024-01-02:
I'm interested on update gcc version, I think that I can't build nav2 for the same reason:
https://github.com/dusty-nv/jetson-containers/issues/336#issuecomment-1823441182
Please, keep this updated!

Thank!!!

mischa-ai, 2024-03-20:
This is my findings so far :

**in short it's not possible**

I tried the same process directly on the jetson nano and also inside the provided container.

I updated the g++ version to 9 and also 11, both fails to build (the ros2_control package and also a bunch of other ros2 packages failed).

I think the combination of Ubuntu 18 (which officially reached end of live and is not supported anymore) and newer g++ and maybe some other dependencies are simply incompatible.

My solution is to use an [osrf container ](https://hub.docker.com/r/osrf/ros/tags) with latest ROS2 build (which does not support the gpu drivers, so I use it only to control the robot).

closed 2024-03-20 by mischa-ai