dusty-nv/jetson-containers #350: How do I containerize my Python application?

sibyjackgrove, 2023-12-11:
Maybe this has been answered before. But I am confused by the instructions in https://github.com/dusty-nv/jetson-containers/blob/master/docs/build.md#building-external-packages 
If I have an application called `dataimputation.py`,  Could you let me know if the following steps are the correct ones if I want to containerize the application by using a Dockerfile?

Build my base container.

`$ ./build.sh --name my_tensorflow tensorflow2 --verbose
`

Create Dockerfile with my  `dataimputation.py` application.
```
FROM my_tensorflow

WORKDIR app
COPY . /app

RUN pip3 install -r requirements.txt

CMD ["python3", "dataimputation.py"]
```

Build my external package using the Dockerfile
`$ ./build.sh --package-dirs=/mypackage dataimpuation
`

sibyjackgrove, 2023-12-13:
I figured it out without using the external packages option. It seems I just need to define a Dockerfile that uses as it's base image the jetson image that I built previously. 

```
FROM tf211_container:r35.4.1

RUN mkdir /app
WORKDIR /app

COPY ./requirements.txt ./
RUN python3 -m pip install --no-cache-dir -r requirements.txt
```
Then use `docker build` instead of `build.sh` to build my custom container
`docker build -t custom_jetson_pacakge:0.0.1 .
`

dusty-nv, 2023-12-13:
Hi @sibyjackgrove, sorry for the delay - yes, that approach you took with making your own Dockerfile is the easiest 👍 

For more complex setups, you can define your own packages in jetson-containers, which become components used in other containers in the build system:  https://github.com/dusty-nv/jetson-containers/blob/master/docs/packages.md

But you can think of jetson-containers as the way to essentially build your desired base container for you to layer your application on top of.

sibyjackgrove, 2023-12-14:
> Hi @sibyjackgrove, sorry for the delay - yes, that approach you took with making your own Dockerfile is the easiest 👍
>
> For more complex setups, you can define your own packages in jetson-containers, [...]

Thank you for confirming my approach @dusty-nv I just wanted to make sure, I was not building custom container's in a sub-optimal way.

gOrca, 2024-05-14:
> dataimputation.py

At what point did you include your 'dataimputation.py' as part of your container? And how did you do this?

closed 2025-02-21 by johnnynunez