dusty-nv/jetson-containers #352: Why is DOCKER_BUILDKIT=0 explicitly set?

ahi-bplus, 2023-12-13:
I would like to build my own packages and need BuildKit. Why is BuildKit explicitly disabled?

https://github.com/dusty-nv/jetson-containers/blob/6f0d65ed3884430d099bf3ffb7777318a6277f9e/jetson_containers/container.py#L105

Thanks for your help.

champred, 2024-07-23:
Apparently BuildKit doesn't allow access to the NVIDIA container runtime which is needed for CUDA libraries. I don't believe there's a workaround, but if there is one it would be nice to know.

dusty-nv, 2024-07-23:
Yes, unlike vanilla `docker build` it does not honor the default docker runtime during container builds.  If you find a workaround or that it is fixed, let me know too!