dusty-nv/jetson-containers #353: l4t_version_compatible() wrongly flags JetPack6 R36 images as compatible on JetPack5 R35 hosts

frostydonut, 2023-12-14:
https://github.com/dusty-nv/jetson-containers/blob/93739890ec4f575c113e0c8cfad76a6e4e1a51d3/jetson_containers/l4t_version.py#L243

">=35" should be "==35" ??

dusty-nv, 2023-12-14:
Ahh thanks @frostydonut , you are correct!  Just fixed this in https://github.com/dusty-nv/jetson-containers/commit/2b178c8ce897556035ecdbf49bb91c1716716ac1

Note that if it pulled any `r36.2.0` images to your system, you should remove them with `docker rmi`, because autotag will defer to your local images first (and for those it doesn't do version checks).  Once you have r36.2.0 removed, it will check dockerhub again for r35 images if you pull the latest jetson-containers that includes the patch above.

frostydonut, 2023-12-14:
> correct! Just fixed this in 2b178c8

Thanks, Dusty! Awesome resource btw. This got me dusting-off an old Xavier
NX and playing with some cool new stuff again. Thanks for bestowing this on
the world! Cheers!

adamwhats, 2023-12-14:
Hi @dusty-nv,  what are the ramifications of running r36.x images on an r35.x host? I've just sat down to try and use the `humble-ros-base-l4t-r36.2.0` on my Xavier NX. My thought (hope) was that I could run the majority of my ROS nodes in that newer humble container, and then put the few which use GPU in to a separate roxy/r35.x container

dusty-nv, 2023-12-14:
@frostydonut no problem, happy to be of service!

@adamwhats the r36 images aren't compatible with r35, not for GPU at least.  Otherwise, yea you can run whatever as long as it was built for aarch64.  And if you *aren't* using GPU for some nodes, then you can just run the upstream OSRF ROS containers for those:   https://github.com/osrf/docker_images   my guess is you would prefer one with a 22.04 base since that has tier-1 support for Humble and their debian packages.