dusty-nv/jetson-containers #354: How do I install Docker compose on Jetson Orin?

sibyjackgrove, 2023-12-14:
I want to run multiple containers on Jetson Orin. It seems docker-compose is the recommended way by docker for this. Is this true, or is there another method more suitable for Jetson Orin devices? If docker-compose is the way to go, could you point me to any documentation on the recommended installation procedure for docker-compose?

I found the documentation from DigitalOcean to be the most easy to follow: https://www.digitalocean.com/community/tutorials/how-to-install-and-use-docker-compose-on-ubuntu-20-04

Could anyone please confirm if the the above instructions will work with Jetson?

dusty-nv, 2023-12-14:
Hi @sibyjackgrove, I've not personally used docker-compose, but I know that @rbonghi and others from the Jetson developer forums have

sibyjackgrove, 2023-12-14:
@rbonghi Could you let me if you have used docker-compose on Jetson devices? Also, could you please point to any documentation on how to install it on Jetson?

dusty-nv, 2023-12-14:
@sibyjackgrove it would not appear different than the normal way to install docker-compose with pip:

* https://collabnix.com/running-docker-compose-on-nvidia-jetson-nano-in-5-minutes/
* https://dev.to/ductapedev/install-docker-compose-on-nvidia-jetson-nano-2oe

rbonghi, 2023-12-15:
Hi all,
It's pretty simple! Follow the official documentation https://docs.docker.com/compose/install/linux/#install-using-the-repository
(if needed, you can also reinstall docker following the official documentation)

sibyjackgrove, 2023-12-15:
> Hi all, It's pretty simple! Follow the official documentation https://docs.docker.com/compose/install/linux/#install-using-the-repository (if needed, you can also reinstall docker following the official documentation)

Thanks for the link. I tried the following command according to the official documentation. 
` sudo apt-get install docker-compose-plugin
`
But I am getting the following error:

`E: Unable to locate package docker-compose-plugin
`
Could you suggest a solution I could try?

johnnynunez, 2023-12-16:
https://hackmd.io/R0tDEhBYTZSaEHpIQYh_Zg

rbonghi, 2023-12-18:
Hi @sibyjackgrove, you can also reinstall the latest Docker from the official documentation. https://docs.docker.com/desktop/install/ubuntu/

Remember after to reinstall nvidia docker

sibyjackgrove, 2023-12-20:
> Hi @sibyjackgrove, you can also reinstall the latest Docker from the official documentation. https://docs.docker.com/desktop/install/ubuntu/
> 
> Remember after to reinstall nvidia docker

Thank you, I will try it out. THough I wish there was a way to do it without having to reinstall the nvidia docker and potentially break something on Jetson.

closed 2025-02-21 by johnnynunez

drew-pi, 2025-06-25:
Using the [link](https://collabnix.com/running-docker-compose-on-nvidia-jetson-nano-in-5-minutes/) provided by @dusty-nv helped, but, for me at least, they were missing one thing. 

After I apt installed pip3 I also needed to run
```
pip3 install --upgrade pip
```
I guess for some reason it didn't download the correct version of pip. But after I ran the above command and then tried 
```
sudo pip3 install docker-compose=="1.29.2"
```

It worked and I got
```
$ docker-compose version
/usr/local/lib/python3.6/dist-packages/paramiko/transport.py:32: CryptographyDeprecationWarning: Python 3.6 is no longer supported by the Python core team. Therefore, support for it is deprecated in cryptography. The next release of cryptography will remove support for Python 3.6.
  from cryptography.hazmat.backends import default_backend
docker-compose version 1.29.2, build unknown
docker-py version: 5.0.3
CPython version: 3.6.9
OpenSSL version: OpenSSL 1.1.1  11 Sep 2018
```