dusty-nv/jetson-containers #358: Troubleshooting jetson-containers on NVIDIA AGX Xavier

HShamimGEHC, 2024-01-08:
Hi - This is my first time using a Jetson Xavier AGX and I am trying to follow the documentations to get a container running.

I had two questions:
1. This AGX Xavier was given to me flashed and JetPack version is 4.1.1 and L4T Version is 31.1.0. On the starter setup, it mentions that JP Version should be >= 5.1 and L4T >= 35.2.1. **Will that be an issue to get containers running?**
2. When I run `./build.sh --list-packages       # list all packages`, I get NameError: name "PROTOBUF_VERSION" is not defined.

dusty-nv, 2024-01-08:
> 1. This AGX Xavier was given to me flashed and JetPack version is 4.1.1 and L4T Version is 31.1.0. On the starter setup, it mentions that JP Version should be >= 5.1 and L4T >= 35.2.1. **Will that be an issue to get containers running?**

Yes, JetPack 4.1.1 is quite ancient by this point, and you should reflash it with JetPack 5.1.2 using SDK Manager

> 2\. When I run `./build.sh --list-packages       # list all packages`, I get NameError: name "PROTOBUF_VERSION" is not defined. **Is protobuf something that comes with later versions of JetPack and L4T or is there another step that I am missing?**

Yes, it is due to your old JetPack.  GPU-accelerated containers weren't really supported on Jetson at all before JetPack 4.4 IIRC

HShamimGEHC, 2024-01-08:
Thank you for the quick reply. 

My follow up is probably not in scope for this repo but I am currently trying to do exactly what you mentioned in your reply for 1 (reflashing with 5.1.2 using SDK Manager). 

**I have a Dell running Ubuntu 20.04 and am trying to connect the Xavier to it using a USBC to USBC connection. Unfortunately, SDK Manager 2.0.0 fails to detect the board. I am not sure what the original intended purpose of this USBC-USBC wire was but do you have any tips/advice regarding this?**

Thank you in advanace.

dusty-nv, 2024-01-08:
@HShamimGEHC double-check that you are connecting to the correct USB-C port for flashing on the AGX, and that it's in recovery mode.  `lsusb` on your PC should show nvidia device pop up.   If you have further issues flashing, I recommend posting it to the jetson developer forums so the linux/hw guys can help ya out.  Good luck!

HShamimGEHC, 2024-01-09:
Thank you. Also last follow up before I close this issue, if I have general questions about some of the containers or how certain things are setup the way they are, is opening an issue the proper way to do so? Or is there some other method recommended?

dusty-nv, 2024-01-09:
> if I have general questions about some of the containers or how certain things are setup the way they are, is opening an issue the proper way to do so?

Oh sure yes @HShamimGEHC, feel free to ask general questions over the GitHub Issues, that is fine 👍