dusty-nv/jetson-containers #359: Error building with l4t-pytorch and ros:humble-ros-base

pfarinha91, 2024-01-09:
Hello, I have a Jetson Orin 64 GB with JetPack 6.0 (l4t 36.2.0), and I'm currently having some issues. I can do a build with l4t-pytorch and ros:humble-desktop or ros:humble-ros-core but not with ros:humble-ros-base.

The commands:

./build.sh --name=my_container l4t-pytorch ros:humble-desktop **OK**

./build.sh --name=my_container l4t-pytorch ros:humble-ros-core **OK**

./build.sh --name=my_container l4t-pytorch ros:humble-ros-base **NOT-OK**

The error:

```
(...)

HEAD is now at fbd4968 Rebase from 'upstream'
The command '/bin/bash -c ./ros2_build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/orin/jetson-containers/jetson_containers/build.py", line 102, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/orin/jetson-containers/jetson_containers/container.py", line 141, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag my_container:l4t-r36.2.0-ros_humble-ros-base --file /home/orin/jetson-containers/packages/ros/Dockerfile.ros2 --build-arg BASE_IMAGE=my_container:l4t-r36.2.0-l4t-pytorch --build-arg ROS_VERSION="humble" --build-arg ROS_PACKAGE="ros_base" /home/orin/jetson-containers/packages/ros 2>&1 | tee /home/orin/jetson-containers/logs/20240109_140654/build/my_container_l4t-r36.2.0-ros_humble-ros-base.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

```

I've ran this multiple times and event tried ros-base with some other variations and the result is always the same.

Am I missing something? Thanks!

dusty-nv, 2024-01-09:
Hi @pfarinha91, please post more of the build log - that snippet doesn't contain the actual error.  I am trying this same build on JetPack 6 now though.

pfarinha91, 2024-01-09:
```
Step 18/23 : RUN ./ros2_build.sh
 ---> Running in 1f2bbefa25b4
ROS2 builder => ROS_DISTRO=humble ROS_PACKAGE=ros_base ROS_ROOT=/opt/ros/humble
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease [270 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease [119 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease [109 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease [110 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy/restricted arm64 Packages [24.2 kB]
Get:6 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 Packages [1,758 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports jammy/multiverse arm64 Packages [224 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports jammy/universe arm64 Packages [17.2 MB]
Get:9 http://ports.ubuntu.com/ubuntu-ports jammy-updates/multiverse arm64 Packages [28.4 kB]
Get:10 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 Packages [1,221 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 Packages [1,396 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports jammy-updates/restricted arm64 Packages [1,068 kB]
Get:13 http://ports.ubuntu.com/ubuntu-ports jammy-backports/main arm64 Packages [49.9 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports jammy-backports/universe arm64 Packages [26.2 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports jammy-security/universe arm64 Packages [965 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports jammy-security/main arm64 Packages [1,131 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports jammy-security/restricted arm64 Packages [1,059 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports jammy-security/multiverse arm64 Packages [24.0 kB]
Fetched 26.8 MB in 7s (3,591 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
lsb-release is already the newest version (11.1.0ubuntu4).
wget is already the newest version (1.21.2-2ubuntu1).
ca-certificates is already the newest version (20230311ubuntu0.22.04.1).
curl is already the newest version (7.81.0-1ubuntu1.15).
The following NEW packages will be installed:
  gnupg2
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,548 B of archives.
After this operation, 52.2 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/universe arm64 gnupg2 all 2.2.27-3ubuntu2.1 [5,548 B]
debconf: delaying package configuration, since apt-utils is not installed
Fetched 5,548 B in 0s (24.2 kB/s)
Selecting previously unselected package gnupg2.
(Reading database ... 42670 files and directories currently installed.)
Preparing to unpack .../gnupg2_2.2.27-3ubuntu2.1_all.deb ...
Unpacking gnupg2 (2.2.27-3ubuntu2.1) ...
Setting up gnupg2 (2.2.27-3ubuntu2.1) ...
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Get:5 http://packages.ros.org/ros2/ubuntu jammy InRelease [4,682 B]
Get:6 http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages [1,387 kB]
Fetched 1,392 kB in 3s (413 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.9ubuntu3).
git is already the newest version (1:2.34.1-1ubuntu1.10).
libpython3-dev is already the newest version (3.10.6-1~22.04).
libpython3-dev set to manually installed.
python3-numpy is already the newest version (1:1.21.5-1ubuntu22.04.1).
python3-setuptools is already the newest version (59.6.0-1.2ubuntu0.22.04.1).
python3-setuptools set to manually installed.
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.4).
The following additional packages will be installed:
  cmake-data dh-elpa-helper docutils-common emacsen-common libbullet3.06
  libcunit1 libjs-jquery-hotkeys libjs-jquery-isonscreen libjs-jquery-metadata
  libjs-jquery-tablesorter libjs-jquery-throttle-debounce libjsoncpp25
  librhash0 libtinyxml2-9 libuv1 python3-argcomplete python3-attr
  python3-catkin-pkg-modules python3-colcon-argcomplete python3-colcon-bash
  python3-colcon-cd python3-colcon-cmake python3-colcon-core
[... 4644 lines cut ...]

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 611a259cf Trimmed the branch to only the ros2cli sub directory
=== src/ros2cli/ros2cli_test_interfaces (git) ===
Cloning into '.'...
Note: switching to 'release/humble/ros2cli_test_interfaces/0.18.7-1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at 27c090d78 Trimmed the branch to only the ros2cli_test_interfaces sub directory
===

dusty-nv, 2024-01-09:
@pfarinha91 weird, I still don't see any actual errors in that log, which makes me wonder if you are running out of disk space or something?

I was able to build this without issue using the same command as you.  I uploaded the container image to `dustynv/test:pytorch-ros-humble-base-r36.2.0` that you can pull if that's helpful.

pfarinha91, 2024-01-10:
Weird, I'm not out of disk space, it must be something else.

But I eventually made it work with ros-core so it's ok. I will close the issue since you couldn't reproduce it!

Thanks for uploading the container.

closed 2024-01-10 by pfarinha91

MilianoAlvez, 2024-02-01:
I get the same exact error. I want to add ROS2 over base image from (https://github.com/lanzani/opencv-cuda-jetson-docker) and I tried it multiple times on both Jetson nano and on my Desktop PC windows 11 but I get the same error no matter what. I have no memory issues at least on my Desktop PC.

[image]

mzahana, 2025-01-24:
I got the same error after trying the following command on Jetson AGX 64GB with JP6.1
```bash
jetson-containers build --name=my_test_image --verbose pytorch torchvision ros:humble-desktop
```