dusty-nv/jetson-containers #361: Is "Track Anything" (TAM) container broken on JetPack 6?

chaos6887, 2024-01-12:
The github autobuilds are failing..

dusty-nv, 2024-01-13:
Ah thanks @chaos6887, addressed this in commit https://github.com/dusty-nv/jetson-containers/commit/9f4df1a159a4d1601da1821a3e55d633b23aa82b - you can try [`dustynv/tam:r36.2.0`](https://hub.docker.com/r/dustynv/tam/tags)

IamShubhamGupto, 2024-01-13:
@dusty-nv with JP6 DP available, I want to start helping build containers as I will be actively using them in the coming weeks. What's the best way to get involved with this repository? Thanks

dusty-nv, 2024-01-13:
@IamShubhamGupto if you want to build containers, I'd follow the System Setup guidelines form here first:  https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md

Then see https://github.com/dusty-nv/jetson-containers/blob/master/docs/build.md

IamShubhamGupto, 2024-01-31:
@dusty-nv thanks for the guides. I have set up my Jetson Orin machine with the above instructions. Is it possible to use a bigger / faster machine for building these containers?

I see that there isn't a JP6 with ros2 iron, realsense, pytorch 2 and tensorRT already available and I will be building it for my project. Let me know if this image is a requirement / what can I build for this repository

dusty-nv, 2024-01-31:
@IamShubhamGupto there is a JetPack cross-compilation container on NGC here:

https://catalog.ngc.nvidia.com/orgs/nvidia/containers/jetpack-linux-aarch64-crosscompile-x86

However each project you build would need support for cross-compilation (like ROS2, librealsense, ect) and I don't attempt it.   I build jetson-containers natively on the 12-core Jetson AGX Orin and that is fast enough for me (i.e. ROS2 takes like an hour to build IIRC...and ROS2 Humble/Iron are also available from the OSRF apt repo for Ubuntu 22.04 on JetPack 6)