dusty-nv/jetson-containers #363: humble-desktop-pytorch-l4t-r35.4.1 - Missing ros_deep_learning

omerts, 2024-01-16:
Hello,

The docker image: `dustynv/ros:humble-desktop-pytorch-l4t-r35.4.1` is missing the ros_deep_learning package:
[image]

Hommus, 2024-03-26:
I came across the same issue when building as follows:
```bash
./build.sh pytorch ros:iron-perception
```

Hommus, 2024-03-26:
@omerts More info here on how to include ros_deep_learning: https://github.com/dusty-nv/jetson-containers/issues/286

Hommus, 2024-03-27:
@dusty-nv Is there a timeline for ros_deep_learning to be included in L4T-r35.4.1 by default? I've been trying to build a ros:iron-perception image with pytorch and ros_deep_learning but I haven't been successful yet.

omerts, 2024-03-27:
> I came across the same issue when building as follows:
> 
> ```shell
> ./build.sh pytorch ros:iron-perception
> ```
> 
> @omerts You should be able to build the image with a previous version of L4T to fix it. Try `l4t-r35.3.1`

Yes it works with 35.3.1, but this is unrelated to the issue which is for 35.4.1.