dusty-nv/jetson-containers #368: `opencv:4.8.1-r36.2.0` container cant run opencv on jetson Xavier

dev10110, 2024-01-25:
Hi @dusty-nv , 

I ve pulled the container `dustynv/opencv:4.8.1-r36.2.0`  on my Jetson Xavier NX. Inside the docker `cv2.getBuildInformation()` indicates that openCV is built with CUDA, but when I try to run any cuda opencv code, I get an error saying the kernel image is not supported. 

I beleive the issue is here: 
https://github.com/dusty-nv/jetson-containers/blob/aac129e6cb927df2bd8a1ae695557d7553d86cb8/packages/opencv/opencv_builder/Dockerfile#L58C1-L58C44
since the `.getBuildInformation` also tells me that it is built for GPU architecture 87, and 72 (necessary for jetson xavier) is not there. 

Could you please advise me on which image I should pull from to avoid this issue? My only requirement is that the image contains Ubuntu 22.04 and Opencv >= 4.7, built with `contrib` modules and with cuda.

dusty-nv, 2024-01-25:
@dev10110 the r36.2.0 containers are for JetPack 6, and Xavier doesn't support JetPack 6.   I am not sure how you would get both 22.04 on Xavier *with* CUDA acceleration, officially at least.

Niproblema, 2024-04-06:
Maan Its a big shame we can't have xavier supported

dev10110, 2024-04-08:
I rebuilt the container with support for the 72 architecture, and it worked for me, but I had to do it manually.

closed 2025-02-21 by johnnynunez