dusty-nv/jetson-containers #369: Cannot run stable-diffusion-webui

RPHllc, 2024-01-26:
When attempting to ./run.sh stable-diffusion-webui:r35.4.1, the code crashes with

`+ docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /home/rp/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb --name aa stable-diffusion-webui:r35.4.1
Python 3.8.10 (default, Nov 22 2023, 10:22:35)
[GCC 9.4.0]
Version: v1.7.0
Commit hash: cf2772fab0af5573da775e7437e6acdca424f26e
Launching Web UI with arguments: --data=/data/models/stable-diffusion --enable-insecure-extension-access --xformers --listen --port=7860
Traceback (most recent call last):
  File "launch.py", line 48, in <module>
    main()
  File "launch.py", line 44, in main
    start()
  File "/opt/stable-diffusion-webui/modules/launch_utils.py", line 460, in start
    import webui
  File "/opt/stable-diffusion-webui/webui.py", line 13, in <module>
    initialize.imports()
  File "/opt/stable-diffusion-webui/modules/initialize.py", line 34, in imports
    shared_init.initialize()
  File "/opt/stable-diffusion-webui/modules/shared_init.py", line 17, in initialize
    from modules import options, shared_options
  File "/opt/stable-diffusion-webui/modules/shared_options.py", line 3, in <module>
    from modules import localization, ui_components, shared_items, shared, interrogate, shared_gradio_themes
  File "/opt/stable-diffusion-webui/modules/localization.py", line 4, in <module>
    from modules import errors, scripts
  File "/opt/stable-diffusion-webui/modules/scripts.py", line 10, in <module>
    from modules import shared, paths, script_callbacks, extensions, script_loading, scripts_postprocessing, errors, timer
  File "/opt/stable-diffusion-webui/modules/script_callbacks.py", line 261, in <module>
    def infotext_pasted_callback(infotext: str, params: dict[str, Any]):
TypeError: 'type' object is not subscriptable`

I investigated a bit and found out that recently the source code was changed from "params: Dict[str, Any]):" to "params: dict[str, Any]):" and Dict is no longer being imported. As "dict" was introduced in Python 3.9, the code from  "https://github.com/AUTOMATIC1111/stable-diffusion-webui)" is no longer compatible with Python 3.8, which we seem to be using in the Dockerfile.

WalkerZJT, 2024-01-30:
so how to deal thsi bug, ths

dusty-nv, 2024-01-30:
@RPHllc @WalkerZJT this is a similar issue to https://github.com/dusty-nv/jetson-containers/issues/364 regarding Python 3.9/3.10 typing hints.   Typically I would just patch these in the dockerfile with sed commands.  Or on JetPack 5, we can lock the stable-diffusion-webui commit to before these changes were made (which given that JetPack 5 is only receiving minor updates/security fixes now that JetPack 6 is out, may be more appropriate)

WalkerZJT, 2024-02-01:
😁ths

RPHllc, 2024-02-01:
I moved on to JetPack 6. There are other version compatibility issues with version 35.4.1, we are probably better off moving on. Thank you.

closed 2024-02-01 by RPHllc

dusty-nv, 2024-02-02:
OK, so I tried patching it for Python 3.8, but there kept being more and more patches required, so I ended up just locking it to v1.6.0 on JetPack 5 in commit https://github.com/dusty-nv/jetson-containers/commit/d8f50633d19b8e460eade442fda19235104c4c7e .  A1111 v1.6.0 was the last version that would work with the minimal patch set.   On JetPack 6, it still uses A1111 master branch and Python 3.10.  

I confirmed that SD-1.5 and SDXL are still working on JetPack 5 with A1111 v1.6.0, and the stable-diffusion-webui containers are all rebuilding now.