dusty-nv/jetson-containers #371: Ugh. Xavier NX Runs out Of Docker Space on "./run.sh $(./autotag text-generation-webui)"

DennisFaucher, 2024-01-29:
My newly installed Xavier NX has 11GB free in /. 
Running "./run.sh $(./autotag text-generation-webui)"
Fills up / to 100% before all layers are downloaded and extracted.
I have an external SD-Card. Can I move /var/lib/docker to the SD-Card? Thanks.

PrinceP, 2024-01-29:
You can.  
https://www.digitalocean.com/community/questions/how-to-move-the-default-var-lib-docker-to-another-directory-for-docker-on-linux

DennisFaucher, 2024-01-29:
Fixed it. The instructions in this blog post were helpful but the syntax (-g) has changed to (--data-root). My working /lib/systemd/system/docker.service has this change:

ExecStart=/usr/bin/dockerd --data-root /media/dennis/64GB_SD_EXT4/docker -H fd:// --containerd=/run/containerd/containerd.sock
#ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock

closed 2024-01-29 by DennisFaucher

DennisFaucher, 2024-02-02:
Oh look. It is all explained here. RTFM Dennis :) 
https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md