dusty-nv/jetson-containers #373: L4T_VERSION=35.4.1  JETPACK_VERSION=5.1.2  CUDA_VERSION=11.4.315

PrinceP, 2024-01-30:
```txt
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.8/dist-packages/mlc_llm/build.py", line 47, in <module>
    main()
  File "/usr/local/lib/python3.8/dist-packages/mlc_llm/build.py", line 43, in main
    core.build_model_from_args(parsed_args)
  File "/usr/local/lib/python3.8/dist-packages/mlc_llm/core.py", line 921, in build_model_from_args
    mod = mod_transform_before_build(mod, param_manager, args, model_config)
  File "/usr/local/lib/python3.8/dist-packages/mlc_llm/core.py", line 648, in mod_transform_before_build
    mod = tvm.transform.Sequential(
  File "/usr/local/lib/python3.8/dist-packages/tvm/ir/transform.py", line 238, in __call__
    return _ffi_transform_api.RunPass(self, mod)
  File "tvm/_ffi/_cython/./packed_func.pxi", line 332, in tvm._ffi._cy3.core.PackedFuncBase.__call__
  File "tvm/_ffi/_cython/./packed_func.pxi", line 263, in tvm._ffi._cy3.core.FuncCall
  File "tvm/_ffi/_cython/./packed_func.pxi", line 252, in tvm._ffi._cy3.core.FuncCall3
  File "tvm/_ffi/_cython/./base.pxi", line 182, in tvm._ffi._cy3.core.CHECK_CALL
  File "/usr/local/lib/python3.8/dist-packages/tvm/_ffi/base.py", line 481, in raise_last_ffi_error
    raise py_err
  File "tvm/_ffi/_cython/./packed_func.pxi", line 56, in tvm._ffi._cy3.core.tvm_callback
  File "/usr/local/lib/python3.8/dist-packages/tvm/contrib/cutlass/build.py", line 1008, in profile_relax_function
    conv2d_profiler = CutlassConv2DProfiler(sm, _get_cutlass_path(), tmp_dir)
  File "/usr/local/lib/python3.8/dist-packages/tvm/contrib/cutlass/gen_conv2d.py", line 186, in __init__
    self.gemm_profiler = CutlassGemmProfiler(sm, cutlass_path, binary_path)
  File "/usr/local/lib/python3.8/dist-packages/tvm/contrib/cutlass/gen_gemm.py", line 197, in __init__
    assert sm in GENERATOR_FUNC_TABLE and sm in DEFAULT_KERNELS, f"sm{sm} not supported yet."
AssertionError: sm72 not supported yet.
```

dusty-nv, 2024-01-30:
@PrinceP I believe that MLC/TVM only supports Ampere GPU architecture (Orin) and newer.   I have it patched to attempt to include SM72, but it doesn't seem to work...  https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/mlc/patches/51fb0f4.diff

Niproblema, 2024-04-06:
@dusty-nv  Are there any models that don't use MLC/TVM that we could run on SM72?

dusty-nv, 2024-04-06:
@Niproblema yes, on SM72 I would just recommend using llama.cpp, exllama, or ollama (we have containers already built for those on JetPack 5)

ttkien2035, 2024-10-01:
@dusty-nv 
I also have the same problem with sm72. If I upgrade jetpack to 6.0, Is this issue resolved?

dusty-nv, 2024-10-01:
@ttkien2035 no, unfortunately sm72 will not get the highest-end kernel support from MLC, vLLM, TensorRT-LLM, ect because it is older GPU generation and does not have those features.  But it is still supported by the likes of llama.cpp exllama and AutoGPTQ.

closed 2024-10-02 by PrinceP