dusty-nv/jetson-containers #374: opencv 4.9.0

johnnynunez, 2024-01-30:
I'm having some problems if I compile with cuda 12.2.
Can you compile it?

johnnynunez, 2024-01-31:
@dusty-nv FYI https://x.com/osanseviero/status/1752631390758158819?s=46

dusty-nv, 2024-01-31:
@johnnynunez yes, just saw about the Llava-1.6 release 😄 

I am scoping the changes needed to support it in my optimized VLM pipeline, my guess is it will mostly have to do with supporting their new 'AnyRes' image grid tiling scheme.   https://llava-vl.github.io/blog/2024-01-30-llava-1-6/

RE: OpenCV, have not tried 4.9, I have been using 4.8.1 built for JetPack 6 and CUDA 12.2 using my [`opencv_builder`](https://github.com/dusty-nv/jetson-containers/tree/master/packages/opencv/opencv_builder) container.

johnnynunez, 2024-02-01:
> @johnnynunez yes, just saw about the Llava-1.6 release 😄
>
> I am scoping the changes needed to support it in my optimized VLM pipeline, my guess is it will mostly have to do with supporting their new [...]

If you try 4.9 tell me

ksvladimir, 2024-03-19:
I'd love to see llava 1.6 support too!

ai-and-i, 2024-03-28:
Awesome, thanks for confirming! I tested llava-1.6 34B, and it's not working - looks like because of this issue: https://github.com/mlc-ai/mlc-llm/issues/1919 (llava-1.6-34b is based on yi-34b and I'm seeing exactly the same error). It seems to be fixed in the latest tvm, so perhaps just need to update it in the docker image? Will try tomorrow.

closed 2024-09-14 by johnnynunez