dusty-nv/jetson-containers #375: OpenCV/Gstreamer error for CSI camera in NanoOwl on Seeed Orin NX

costasvav, 2024-02-01:
Attempting to run the NanoOWL container on a Seeed Orin NX 8GB (J401 carrier board) running Jetpack 5.1.2.  This is a clean install.

I followed the steps from [here](https://www.jetson-ai-lab.com/vit/tutorial_nanoowl.html), but I get the following errors:

```
user@jetson-nx:~/jetson-containers$ ./run.sh $(./autotag nanoowl)
Namespace(disable=[''], output='/tmp/autotag', packages=['nanoowl'], prefer=['local', 'registry', 'build'], quiet=False, user='dustynv', verbose=False)
-- L4T_VERSION=35.4.1  JETPACK_VERSION=5.1.2  CUDA_VERSION=11.4.315
-- Finding compatible container image for ['nanoowl']
[sudo] password for user:
dustynv/nanoowl:r35.3.1
+ sudo docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /home/user/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb --device /dev/video0 dustynv/nanoowl:r35.3.1
root@jetson-nx:/opt/nanoowl# cd examples/tree_demo
root@jetson-nx:/opt/nanoowl/examples/tree_demo# python3 tree_demo.py ../../data/owl_image_encoder_patch32.engine
/usr/local/lib/python3.8/dist-packages/transformers/utils/hub.py:123: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
/usr/local/lib/python3.8/dist-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /opt/pytorch/pytorch/aten/src/ATen/native/TensorShape.cpp:3483.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
[02/01/2024-15:05:06] [TRT] [W] Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
100%|███████████████████████████████████████| 338M/338M [00:20<00:00, 17.2MiB/s]
INFO:root:Opening camera.
[ WARN:0] global /opt/opencv/modules/videoio/src/cap_gstreamer.cpp (1760) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0] global /opt/opencv/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /opt/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
INFO:root:Loading predictor.
======== Running on http://0.0.0.0:7860 ========
(Press CTRL+C to quit)
[ WARN:1] global /opt/opencv/modules/videoio/src/cap_v4l.cpp (1004) tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.
```

Camera feed works perfectly outside the container.  Thoughts?

dusty-nv, 2024-02-01:
Hmm not sure...this NanoOWL project/container uses different video I/O through OpenCV than I typically use (through jetson_utils)

@tokk-nv @jaybdub can you take a look pls?

burningion, 2024-02-08:
Have the same experience, it seems: 

`Embedded video playback halted; module v4l2src0 reported: Internal data stream error.` is where things go wrong.

Above this error though, I see the following errors: 

```
sh: 1: lsmod: not found
sh: 1: modprobe: not found
```

Wondering where this may come from, as I don't see it in the nanoowl Dockerfile, or in any of the other container dependencies. (Done via text search for "lsmod" and "modprobe" in this repo.)

I'm trying to use a USB webcam (C920), or a Realsense (455), and can get neither to work. Tried installing DeepStream thinking maybe my firmware was missing something. (On Jetpack 6.)

Here's the relevant output around the gstreamer error for me:

```
INFO:root:Opening camera.

(gst-plugin-scanner:36): GLib-GObject-WARNING **: 20:42:12.951: cannot register existing type 'GstRtpSrc'

(gst-plugin-scanner:36): GLib-GObject-CRITICAL **: 20:42:12.951: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:36): GLib-CRITICAL **: 20:42:12.951: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:36): GStreamer-CRITICAL **: 20:42:12.951: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed

(gst-plugin-scanner:36): GLib-GObject-WARNING **: 20:42:12.951: cannot register existing type 'GstRtpSink'

(gst-plugin-scanner:36): GLib-GObject-CRITICAL **: 20:42:12.951: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:36): GLib-CRITICAL **: 20:42:12.951: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:36): GStreamer-CRITICAL **: 20:42:12.951: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sh: 1: lsmod: not found
sh: 1: modprobe: not found
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 266)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 339)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 355)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 368)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
[ WARN:0@52.575] global cap_gstreamer.cpp:2784 handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Internal data stream error.
[ WARN:0@52.576] global cap_gstreamer.cpp:1679 open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@52.576] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0@52.651] global cap_v4l.cpp:982 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ERROR:0@52.661] global obsensor_uvc_stream_channel.cpp:156 getStreamChannelGroup Camera index out of range
```

burningion, 2024-02-08:
Looking further, error appears to be in OpenCV dependency:

```
$ ./run.sh $(./autotag opencv) /bin/bash 
.... downloads container... 
... in container:
> $ pip install ipython
> $ ipython
>> import cv2
>> a = cv2.VideoCapture(0)
(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.318: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvegltransform.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.319: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvivafilter.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.330: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideosinks.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.334: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvarguscamerasrc.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.338: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideosink.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.340: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvv4l2camerasrc.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.341: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvcompositor.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.343: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvideo4linux2.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.346: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglstreamsrc.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.350: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvjpeg.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.351: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnvvidconv.so': libEGL.so.1: cannot open shared object file: No such file or directory

(gst-plugin-scanner:44): GStreamer-WARNING **: 21:02:32.351: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstnveglglessink.so': libEGL.so.1: cannot open shared object file: No such file or directory
[ WARN:0@8.215] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0@8.495] global cap_v4l.cpp:982 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ERROR:0@8.697] global obsensor_uvc_stream_channel.cpp:156 getStreamChannelGroup Camera index out of range
```

burningion, 2024-02-08:
I think the `jetson_utils` package may be broken too:

```
./run.sh $(./autotag jetson-utils) /bin/bash
Unable to find image 'dustynv/jetson-utils:r36.2.0' locally
r36.2.0: Pulling from dustynv/jetson-utils
... pulls the image ...
>  pip install ipython
>  ipython
>> import jetson_utils
>> b =  jetson_utils.videoSource("/dev/video0") # this is my Realsense
(gst-plugin-scanner:46): GLib-GObject-WARNING **: 21:33:40.555: cannot register existing type 'GstRtpSrc'

(gst-plugin-scanner:46): GLib-GObject-CRITICAL **: 21:33:40.555: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:46): GLib-CRITICAL **: 21:33:40.555: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:46): GStreamer-CRITICAL **: 21:33:40.555: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed

(gst-plugin-scanner:46): GLib-GObject-WARNING **: 21:33:40.555: cannot register existing type 'GstRtpSink'

(gst-plugin-scanner:46): GLib-GObject-CRITICAL **: 21:33:40.555: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:46): GLib-CRITICAL **: 21:33:40.555: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:46): GStreamer-CRITICAL **: 21:33:40.555: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
sh: 1: lsmod: not found
sh: 1: modprobe: not found
(Argus) Error EndOfFile: Unexpected error in reading socket (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 266)
(Argus) Error EndOfFile: Receive worker failure, notifying 1 waiting threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 339)
(Argus) Error InvalidState: Argus client is exiting with 1 outstanding client threads (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadCore(), line 355)
(Argus) Error EndOfFile: Receiving thread terminated with error (in src/rpc/socket/client/ClientSocketManager.cpp, function recvThreadWrapper(), line 368)
(Argus) Error EndOfFile: Client thread received an error from socket (in src/rpc/socket/client/ClientSocketManager.cpp, function send(), line 145)
(Argus) Error EndOfFile:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function dispatch(), line 91)
[gstreamer] initialized gstreamer, version 1.20.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Intel(R) RealSense(TM) Depth Ca
[gstreamer] v4l2-proplist, device.path=(string)/dev/video4, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Intel\(R\)\ RealSense\(TM\)\ Depth\ Ca", v4l2.device.bus_info=(string)usb-3610000.usb-1.1, v4l2.device.version=(uint)331642, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: Intel(R) RealSense(TM) Depth Ca
[gstreamer] v4l2-proplist, device.path=(string)/dev/video2, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Intel\(R\)\ RealSense\(TM\)\ Depth\ Ca", v4l2.device.bus_info=(string)usb-3610000.usb-1.1, v4l2.device.version=(uint)331642, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found v4l2 device: HD Pro Webcam C920
[gstreamer] v4l2-proplist, device.path=(string)/dev/video6, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"HD\ Pro\ Webcam\ C920", v4l2.device.bus_info=(string)usb-3610000.usb-2.2, v4l2.device.version=(uint)331642, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- could not find v4l2 device /dev/video0
[gstreamer] gstCamera -- device discovery failed, but /dev/video0 exists
[gstreamer]              support for compressed formats is disabled
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720, framerate=30/1 ! jpegdec name=decoder ! video/x-raw ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      MJPEG
  -- codecType:  cpu
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
------------------------------------------------
>>  c = jetson_utils.videoSource("/dev/video3") # this is my C920
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video3
[gstreamer] gstCamera -- found v4l2 device: Intel(R) RealSense(TM) Depth Ca
[gstreamer] v4l2-proplist, device.path=(string)/dev/video4, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Intel\(R\)\ RealSense\(TM\)\ Depth\ Ca", v4l2.device.bus_info=(string)usb-3610000.usb-1.1, v4l2.device.version=(uint)331642, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[... 14 lines cut ...]
  -- URI: v4l2:///dev/video3
[... cut ...]
  -- flipMethod: none
------------------------------------------------
>>> b = c.Capture()
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstCamera failed to set pipeline state to PLAYING (error 0)
---------------------------------------------------------------------------
Exception                                 Traceback (most recent call last)
Cell In[12], line 1
----> 1 b = c.Capture()

Exception: jetson.utils -- videoSource failed to capture image
```

bblumberg, 2024-02-09:
Yes, the realsense shows up as 3 devices: one for depth, one for ir and one for rgb (maybe in that order???)

closed 2025-02-21 by johnnynunez