dusty-nv/jetson-containers #382: Where is the docker container data mounted to on my system?

jacobhq, 2024-02-07:
In run.sh, I see:
```
--volume $ROOT/data:/data \
```
on line 54, however when I run `echo $ROOT`, it outputs an empty line.

My end goal is to have this data be stored on an external drive, however I can't find this data folder on my SD card at the moment.

Also just wanted to mention how helpful I find this project - it's great!

Any help would be appreciated. Thanks!

dusty-nv, 2024-02-15:
Hi @jacobhq, sorry for the delay - the `$ROOT` variable gets set inside the run.sh script to the directory in which run.sh is located in (your jetson-containers directory)

https://github.com/dusty-nv/jetson-containers/blob/2d6187b00eaad34a4a51bf1e088baf4a600faa09/run.sh#L4

(the `$ROOT` variable won't persist outside of run.sh, which is maybe why you can't print it?)

Anyways, yes the standard location for the model/data cache is in your jetson-containers/data directory, and it would be easiest to just clone jetson-containers to your SD card - but you can change that data directory to wherever you want on your device (but it should always be mounted to the same place inside the container under `/data`)

run.sh is just a helper script and will print out the `docker run` command that it invokes - you can either edit the script to change this command, or ditch run.sh completely and just run your own `docker run` command (both ways are shown in the package readme's in the jetson-container docs)

Also, just to point it out - your jetson-containers repo does not store the containers themselves (those container images are stored by the docker daemon elsewhere, in what's known as the [docker data root](https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#relocating-docker-data-root)).  Both your jetson-containers repo and docker data root can be moved around independently.

jacobhq, 2024-02-15:
Ok, thanks - that makes sense! So my understanding is that the data directory is not stored in the docker data root. I've moved the docker root to an external SSD following the steps outlined in [setup.md](https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#relocating-docker-data-root). To do the same with the data directory, should I just clone the repo to the SSD or is there a better way (like just changing the location of the data directory mounted to the container)? If so, how would I do that?

dusty-nv, 2024-02-15:
You can literally just move (or re-clone) your jetson-containers repo to your desired location and start the container again, and by virtue of that $ROOT variable it will start the container next time to mount `/data` to your updated jetson-containers location.

jacobhq, 2024-02-15:
Great, thank you!

closed 2024-02-15 by jacobhq