dusty-nv/jetson-containers #388: extend.sh or equivalent to add packages to existing built container

brianlmerritt, 2024-02-16:
I decided to try the mother of all builds (ros:humble-desktop, plus all the AI goodness etc), but some packages such as onnx failed.

This made me think - go back to ros:humble-desktop with pytorch and add the AI goodies later.

Not sure if this is me not knowing how to operate jetson-containers, or is there a missing ability to do something like

`./extend.sh --name=my_jetson_container_name llava transformers etc etc`

I can add all my own stuff into the built container using Dockerfile or compose, but adding jetson-container packages seems less possible.

brianlmerritt, 2024-02-16:
It's probably something I am doing, but if I run 

`./build.sh --name=ros_limo_base:latest pytorch ros:humble-desktop` I am not finding any container with that name.

What I get with docker ps -a is

`18dad6459db7   504f5e673948   "/bin/bash -c 'pytho…"   4 minutes ago       Exited (1) 4 minutes ago                 busy_driscoll`

So the equivalent of ./extend.sh would be `./build.sh --base=my_jetson_container_name` but it gives the error

`pull access denied for my_jetson_container_name, repository does not exist or may require 'docker login': denied: requested access to the resource is denied`

brianlmerritt, 2024-02-16:
Maybe something is going wrong with the ros:humble-desktop build?

I get 
```
GPU devices: 1
loaded test image from /tmp/test_0.jpg  (560, 700, 3)  uint8
OpenCV OK

-- Testing container ros_limo_base:l4t-r36.2.0 (ros:humble-desktop/test_ros2.sh)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/blm/docker/jetson-containers/packages/ros:/test \
--volume /home/blm/docker/jetson-containers/data:/data \
--workdir /test \
ros_limo_base:l4t-r36.2.0 \
/bin/bash -c '/bin/bash test_ros2.sh' \
2>&1 | tee /home/blm/docker/jetson-containers/logs/20240216_113356/test/ros_limo_base_l4t-r36.2.0_test_ros2.sh.txt; exit ${PIPESTATUS[0]}

sourcing   /opt/ros/humble/install/setup.bash
ROS_DISTRO humble
ROS_ROOT   /opt/ros/humble
getting ROS version -
```

Followed by a whole bunch of packages, but the tag container output is missing.

```
    print(f"-- Tagging container {source} -> {target}")
    print(f"{cmd}\n")
```

brianlmerritt, 2024-02-16:
All the "tagged stuff" is way back, buried by the testing afterwards.  Closing now

closed 2024-02-16 by brianlmerritt

dusty-nv, 2024-02-16:
OK gotcha @brianlmerritt - I think you came to the right idea with the `--base` argument to build.sh to change it to your own base container image (although you need to specify the full tag).   There is also `--skip-packages` argument that will disable packages in the build.