dusty-nv/jetson-containers #389: Successfully built containers, but fails Getting Started

bryanhughes, 2024-02-16:
I am following the instructions, though I build the containers first without any errors:

```
$ ./build.sh --name=my_container pytorch transformers ros:humble-desktop
```

I then ran the following:

```
pip3 install -r requirements.txt
./run.sh $(./autotag l4t-pytorch)
```

No issues with the `pip3 install`, but the `./run.sh $(./autotag l4t-pytorch)` fails with `docker: unknown server OS: .` :

```
$ ./run.sh $(./autotag l4t-pytorch)
Namespace(packages=['l4t-pytorch'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.2.0  JETPACK_VERSION=6.0  CUDA_VERSION=12.2.140
-- Finding compatible container image for ['l4t-pytorch']

Found compatible container dustynv/l4t-pytorch:r36.2.0 (2023-12-14, 7.3GB) - would you like to pull it? [Y/n] Y
dustynv/l4t-pytorch:r36.2.0
localuser:root being added to access control list
+ docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /home/bryan/git/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix -v /tmp/.docker.xauth:/tmp/.docker.xauth -e XAUTHORITY=/tmp/.docker.xauth --device /dev/video0 --device /dev/video1 dustynv/l4t-pytorch:r36.2.0
docker: unknown server OS: .
See 'docker run --help'.
```

I have a brand new Jetson AGX Orin 64GB Developer Kit that has been updated following the online guides. Sadly my **docker-fu** is weak.

bryanhughes, 2024-02-16:
This happens when I try to run any of the containers: `docker: unknown server OS: .`

dusty-nv, 2024-02-16:
Hi @bryanhughes, can you post the output of `docker info` ?  This is what mine looks like on JetPack 6:

```bash
sudo docker info
Client: Docker Engine - Community
 Version:    24.0.7
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.11.2
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.21.0
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 2
  Running: 1
  Paused: 0
  Stopped: 1
 Images: 2707
 Server Version: 24.0.7
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 nvidia runc
 Default Runtime: nvidia
 Init Binary: docker-init
 containerd version: d8f198a4ed8892c764191ef7b3b06d8a2eeb5c7f
 runc version: v1.1.10-0-g18a0cb0
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 5.15.122-tegra
 Operating System: Ubuntu 22.04.3 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 12
 Total Memory: 61.37GiB
 Name: jao-60
 ID: ccf2066a-b97e-40a8-b4a1-b56147c3dadb
 Docker Root Dir: /mnt/NVME/docker/data/JetPack_6.0_DP_b32/data
 Debug Mode: false
 Username: dustynv
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```

Are you able to start the base l4t-jetpack container like this:

```bash
sudo docker run -it --rm --net=host --runtime nvidia nvcr.io/nvidia/l4t-jetpack:r35.3.1
```

bryanhughes, 2024-02-16:
Hi @dusty-nv, here is the output of `docker info`

```
$ sudo docker info
Client: Docker Engine - Community
 Version:    25.0.3
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.12.1
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.24.5
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 2
  Running: 1
  Paused: 0
  Stopped: 1
 Images: 111
 Server Version: 25.0.3
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: systemd
 Cgroup Version: 2
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 nvidia runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: ae07eda36dd25f8a1b98dfbf587313b99c0190bb
 runc version: v1.1.12-0-g51d5e94
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
  cgroupns
 Kernel Version: 5.15.122-tegra
 Operating System: Ubuntu 22.04.4 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 8
 Total Memory: 61.37GiB
 Name: mimzy-jetson
 ID: 7a6f660d-b63b-416b-b220-dae947e00ed5
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
```
When I run the docker command directly, it correctly pulls everything and gets me to the command line prompt in the contain.

But when I try to use the `./run.sh`, that still fails. 

Thanks again for all your help! This is a very active repository and very impressive.

bryanhughes, 2024-02-16:
UPDATE: I am to run the docker container directly, but still facing issues. I decided to play around with the `nanoowl` container. I ran:

```
sudo docker run --runtime nvidia -it --rm --network=host dustynv/nanoowl:r36.2.0
```

From within the container I did the following:

```
cd /opt/nanoowl/examples/tree_demo/
python3 tree_demo.py ../../data/owl_image_encoder_patch32.engine
```
This is the output:

```
# python3 tree_demo.py ../../data/owl_image_encoder_patch32.engine
/usr/local/lib/python3.10/dist-packages/transformers/utils/hub.py:123: FutureWarning: Using `TRANSFORMERS_CACHE` is deprecated and will be removed in v5 of Transformers. Use `HF_HOME` instead.
  warnings.warn(
/usr/local/lib/python3.10/dist-packages/torch/functional.py:504: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at /tmp/pytorch/aten/src/ATen/native/TensorShape.cpp:3526.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
[02/16/2024-23:45:50] [TRT] [W] Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
100%|███████████████████████████████████████| 338M/338M [00:06<00:00, 52.4MiB/s]
INFO:root:Opening camera.

(gst-plugin-scanner:51): GLib-GObject-WARNING **: 23:46:03.130: cannot register existing type 'GstRtpSrc'

(gst-plugin-scanner:51): GLib-GObject-CRITICAL **: 23:46:03.131: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:51): GLib-CRITICAL **: 23:46:03.131: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:51): GStreamer-CRITICAL **: 23:46:03.131: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed

(gst-plugin-scanner:51): GLib-GObject-WARNING **: 23:46:03.131: cannot register existing type 'GstRtpSink'

(gst-plugin-scanner:51): GLib-GObject-CRITICAL **: 23:46:03.131: g_type_add_interface_static: assertion 'G_TYPE_IS_INSTANTIATABLE (instance_type)' failed

(gst-plugin-scanner:51): GLib-CRITICAL **: 23:46:03.131: g_once_init_leave: assertion 'result != 0' failed

(gst-plugin-scanner:51): GStreamer-CRITICAL **: 23:46:03.131: gst_element_register: assertion 'g_type_is_a (type, GST_TYPE_ELEMENT)' failed
(Argus) Error FileOperationFailed: Connecting to nvargus-daemon failed: No such file or directory (in src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 204)
(Argus) Error FileOperationFailed: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 106)
sh: 1: lsmod: not found
sh: 1: modprobe: not found
[ WARN:0@19.508] global cap_gstreamer.cpp:2784 handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module v4l2src0 reported: Cannot identify device '/dev/video0'.
[ WARN:0@19.508] global cap_gstreamer.cpp:1679 open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@19.508] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ WARN:0@19.508] global cap_v4l.cpp:982 open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[ WARN:0@19.513] global obsensor_stream_channel_v4l2.cpp:82 xioctl ioctl: fd=-1, req=-2140645888
[ WARN:0@19.513] global obsensor_stream_channel_v4l2.cpp:138 queryUvcDeviceInfoList ioctl error return: 9
[ WARN:0@19.513] global obsensor_stream_channel_v4l2.cpp:82 xioctl ioctl: fd=-1, req=-2140645888
[ WARN:0@19.513] global obsensor_stream_channel_v4l2.cpp:138 queryUvcDeviceInfoList ioctl error return: 9
[ERROR:0@19.513] global obsensor_uvc_stream_channel.cpp:156 getStreamChannelGroup Camera index out of range
INFO:root:Loading predictor.
======== Running on http://0.0.0.0:7860 ========
(Press CTRL+C to quit)
ERROR:aiohttp.server:Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/aiohttp/web_protocol.py", line 350, in data_received
    messages, upgraded, tail = self._request_parser.feed_data(data)
  File "aiohttp/_http_parser.pyx", line 557, in aiohttp._http_parser.HttpParser.feed_data
aiohttp.http_exceptions.BadStatusLine: 400, message:
  Invalid method encountered:

    b'\x16\x03\x01\x02z\x01'
      ^
INFO:aiohttp.access:127.0.0.1 [16/Feb/2024:23:46:33 +0000] "UNKNOWN / HTTP/1.0" 400 222 "-" "-"
INFO:root:handle_index_get
INFO:aiohttp.access:127.0.0.1 [16/Feb/2024:23:46:35 +0000] "GET / HTTP/1.1" 200 236 "-" "Mozilla/5.0 (X11; Linux aarch64; rv:122.0) Gecko/20100101 Firefox/122.0"
INFO:root:Websocket connected.
INFO:aiohttp.access:127.0.0.1 [16/Feb/2024:23:46:36 +0000] "GET /favicon.ico HTTP/1.1" 404 173 "http://0.0.0.0:7860/" "Mozilla/5.0 (X11; Linux aarch64; rv:122.0) Gecko/20100101 Firefox/122.0"
```
When I navigate to `http://0.0.0.0:7860/` it shows the page, but where the camera stream should be there is nothing. I have a Logitech USB camera connected and have confirmed it is present on the host with `v4l2-ctl --list-devices`. Unfortunately this command is not available in the container and not surprisingly, when I try to apt install it, it can not find the package.

bryanhughes, 2024-02-17:
UPDATE:
Figured it out. Need to add `--device /dev/video0` to the docker command line.

closed 2024-02-17 by bryanhughes