dusty-nv/jetson-containers #391: Missing `torchvision` for JP60 images running CUDA 12.2

IamShubhamGupto, 2024-02-16:
Hey Dusty,

I am in the process of building my own container and have used your `build.sh` to create a container for Jetson Orin nano running JP60 preview. However inside the container
I have pytorch version `2.1.0` however it did not install `torchvision`. When I try to manually pip install it, it is incompatible as pytorch hasn't released support for CUDA 12.2.

I verified I am running cuda 12.2 in the container using `nvcc --version`. 

Installing torch and torchvision with cuda 12.1 
`pip install torch==2.1.0 torchvision==0.16.0 torchaudio==2.1.0 --index-url https://download.pytorch.org/whl/cu121` gives `torch.cuda.is_available()` as False 

The nightly builds fails to resolve a version 
`pip3 install --pre torch torchvision torchaudio --index-url https://download.pytorch.org/whl/nightly/cu121`

Is there a way to downgrade cuda to 12.1 in the container or provide support for torchvision with cuda 12.2 ?

Thank you!

dusty-nv, 2024-02-16:
Hi @IamShubhamGupto, you can use my l4t-pytorch container which includes PyTorch, torchvision, and torchaudio and has container image for it up on DockerHub:

* https://github.com/dusty-nv/jetson-containers/tree/master/packages/l4t/l4t-pytorch
* [dustynv/l4t-pytorch:r36.2.0](https://hub.docker.com/r/dustynv/l4t-pytorch/tags)

Or if you are building your own container using build.sh, just add `torchvision` to your list of packages.   Any aarch64 wheels from PyPi or pytorch.org won't be built with CUDA enabled, which is why I build them from source.

IamShubhamGupto, 2024-02-16:
hey @dusty-nv thanks for the quick response, I will give that a try and post an update. I wasn't aware I had to compile `torchvision` manually. 

Thanks

IamShubhamGupto, 2024-02-17:
update, adding torchvision to the required packages solved the issue

closed 2024-02-17 by IamShubhamGupto