dusty-nv/jetson-containers #392: Error parsing YAML: found unexpected ':'

KnechtNoobrecht, 2024-02-17:
I am trying to build a container to run ROS2 Foxy on my Jetson Nano with Jetpack 4.6.4.
So i downloaded the repo, installed all pip dependencies, set the docker default runtime to nvidia and verified that setting.
Now when i try to run `./build.sh --list-packages` i get the following error:
```
Namespace(base='', build_flags='', list_packages=False, logs='', multiple=False, name='', no_github_api=False, package_dirs=[''], packages=['ros:foxy-ros-base'], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''], test_only=[''], verbose=False)
-- L4T_VERSION=32.7.4
-- JETPACK_VERSION=4.6.4
-- CUDA_VERSION=10.2.300
-- LSB_RELEASE=18.04 (bionic)
-- Error parsing YAML from /home/jetson/jetson-containers/packages/tensorflow/Dockerfile:  while scanning a plain scalar
  in "<unicode string>", line 3, column 50:
     ... cudnn, tensorrt, python, numpy, protobuf:cpp]
                                         ^
found unexpected ':'
  in "<unicode string>", line 3, column 58:
     ... ensorrt, python, numpy, protobuf:cpp]
                                         ^
Please check http://pyyaml.org/wiki/YAMLColonInFlowContext for details.

```

dusty-nv, 2024-02-18:
Hi @KnechtNoobrecht, this is strange, I just ran the same command without error on JetPack 4.6.

What does `pip3 show pyyaml` report for you?

```bash
pip3 show pyyaml
Name: PyYAML
Version: 6.0.1
Summary: YAML parser and emitter for Python
Home-page: https://pyyaml.org/
Author: Kirill Simonov
Author-email: xi@resolvent.net
License: MIT
Location: /home/nvidia/.local/lib/python3.6/site-packages
Requires:
```

KnechtNoobrecht, 2024-02-18:
Thanks for your answer, @dusty-nv.
`pip3 show pyyaml` reports the following:
```
Name: PyYAML
Version: 3.12
Summary: YAML parser and emitter for Python
Home-page: http://pyyaml.org/wiki/PyYAML
Author: Kirill Simonov
Author-email: xi@resolvent.net
License: MIT
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: rosdep, rosdep-modules, rosinstall-generator, rospkg, rospkg-modules
```
I see a huge version difference, but the latest version available in the bionic apt repositories is 3.12.

dusty-nv, 2024-02-18:
Interesting... during the `pip3 install -r requirements` command from the [System Setup](https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#clone-the-repo), it should have installed a similar version of PyYAML to mine on your system.  Can you run that, or `pip3 install --upgrade pyyaml` ?

KnechtNoobrecht, 2024-02-18:
Running `pip3 install -r requirements.txt` results in following output:
 `Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from -r requirements.txt (line 2)) (3.12)`
 But `pip3 install --upgrade pyyaml` upgraded the version to 6.0.1. Maybe that's something one could update in the requirements.txt, to specifically demand version 6.0.1. 
 Thanks for your help!

dusty-nv, 2024-02-18:
Thanks yes @KnechtNoobrecht, that must be how this error occurred for you - I will update requirements.txt to install `pyyaml>6`

closed 2024-02-19 by KnechtNoobrecht