dusty-nv/jetson-containers #398: Realsense package on L4T.r36.2 doesn't build

brianlmerritt, 2024-02-22:
I find that the pyrealsense doesn't get created on the current development branch of librealsense, so the test and build fails. 

I found it now works fine on `ARG LIBREALSENSE_VERSION=master`

brianlmerritt, 2024-02-22:
Also getting a rather weird udev error now 

The build is `./build.sh --name=my_container pytorch transformers opencv realsense` and it fails on the build line `cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/ && \`

When I run the container without this, /etc/udev doesn't exist.

I will dig into this a bit further once the container rebuilds (I did a mkdir -p /etc/udev/rules.d just to see what happens) and will also circle back just the first images and see if any of them give me a clue.

dusty-nv, 2024-02-22:
Thanks @brianlmerritt, appreciate it - I don't have an actual realsense to test this against, but will change it to build librealsense `master` on JP6 and install `udev` from apt 👍

dusty-nv, 2024-02-22:
@brianlmerritt updated this in commit https://github.com/dusty-nv/jetson-containers/commit/25b635529a685d2cbb9ed23c2901ef5c05d90ec6 , the containers are building now

apparently I had pinned it to `development` branch previously because of https://github.com/IntelRealSense/librealsense/issues/11931 and a new version of librealsense has still not been released yet, but hopefully those fixes for that previous issue are in master branch by now.

brianlmerritt, 2024-02-22:
I'll try to get an older Jetpack image on my Nano and Xavier and check it still works.

brianlmerritt, 2024-02-23:
OK. ps - I found a way to install cartographer, navigator2 and a few more sensors here without losing cuda support or overwriting opencv.  In case any of the code is useful.

https://gist.github.com/brianlmerritt/68e11e51d3ad219ad19a94f0916e2bc7

bwmcin, 2024-06-10:
I had a similar issue with both r36.2 and r36.3.  I searched and couldn't find pyrealsense2 anywhere in the image.   I deleted the -DPYTHON_INSTALL_DIR line in the dockerfile and built the image.   Importing pyrealsense2 worked (and the package is located in /usr/local/lib/python3.10/dist-packages/pyrealsense2/).

singh-jayant, 2024-07-21:
Hi, `jetson-containers build realsense` seems to be failing.

```
Step 6/6 : RUN python3 -c 'import pyrealsense2'
 ---> Running in 4daccbc5d860
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2'
The command '/bin/sh -c python3 -c 'import pyrealsense2'' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/orin_ssd/dockerfiles/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/orin_ssd/dockerfiles/jetson-containers/jetson_containers/container.py", line 143, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag realsense:r36.3.0-realsense --file /orin_ssd/dockerfiles/jetson-containers/packages/hardware/realsense/Dockerfile --build-arg BASE_IMAGE=realsense:r36.3.0-cmake /orin_ssd/dockerfiles/jetson-containers/packages/hardware/realsense 2>&1 | tee /orin_ssd/dockerfiles/jetson-containers/logs/20240722_004843/build/realsense_r36.3.0-realsense.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

Huizerd, 2024-10-25:
https://github.com/dusty-nv/jetson-containers/issues/281#issuecomment-1874942046 fixed it for me on JP 6.0

closed 2025-02-21 by johnnynunez