dusty-nv/jetson-containers #399: opencv_builder usage - installing opencv_contribs

gtingstad94, 2024-02-24:
I am trying to install opencv with cuda support and the additional cuda-enabled modules. I have been able to successfully install the opencv image and verify that cuda is enabled, but it seems that the additional opencv-contrib packages that leverage gpu acceleration are not baked into the default docker image? Upon installation, and when running `cv2.getBuildInformation()`, I see a long list of to-be-built packages.

I see that the Dockerfile in opencv_builder includes a line to bundle in the additional opencv-contrib files, but as far as I have been able to tell, this configuration is not used during the "vanilla" installation of the opencv image. I have tried building with `./build.sh opencv_builder`, as recommended [here](https://github.com/dusty-nv/jetson-containers/tree/master/packages/opencv/opencv_builder), but I receive a KeyError: "couldn't find package: opencv_builder". 

Apologies if the answer is clear to some. I'm a fresher with Docker.

dusty-nv, 2024-02-24:
Hi @gtingstad94, the opencv-builder container outputs a tarball of opencv deb's, which I then download/extract/install in the opencv container.  If you extract the aarch64-libs deb in there, it has contrib libraries in it which get installed (like the cuda ones)

Also I test during cv2.cuda during the builds, and it wouldn't pass if it wasn't installed?  https://github.com/dusty-nv/jetson-containers/blob/master/packages/opencv/test.py

dusty-nv, 2024-02-24:
Also, it's `opencv:4.8.1-builder` or `opencv:4.5.0-builder` for the name of the builder containers (and you can list them all with `./build.sh --list-packages` or `./build.sh --show-packages opencv*`)

gtingstad94, 2024-02-25:
Hi @dusty-nv, thank you for getting back to me. I have done a bit more digging and realize I have made a mistake. The additional openCV packages do install with the jetson-containers opencv module and I was simply trying to access them incorrectly. 

I appreciate your support!

closed 2025-02-21 by johnnynunez