dusty-nv/jetson-containers #400: (Docker not running) xauth:  file /tmp/.docker.xauth does not exist

rajeshroy402, 2024-02-25:
Hi, I have a problem after I installed python 3.11 recently. I still have python 3.10 in my system as default python3

Now that I try to run my docker, it fails. Sharing terminal logs below: 

` ` ` 

Namespace(packages=['local_llm'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.2.0  JETPACK_VERSION=6.0  CUDA_VERSION=12.2.140
-- Finding compatible container image for ['local_llm']
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/rajesh/jetson-containers/jetson_containers/tag.py", line 58, in <module>
    image = find_container(args.packages[0], prefer_sources=args.prefer, disable_sources=args.disable, user=args.user, quiet=args.quiet)
  File "/home/rajesh/jetson-containers/jetson_containers/container.py", line 501, in find_container
    local_images = find_local_containers(package, **kwargs)
  File "/home/rajesh/jetson-containers/jetson_containers/container.py", line 413, in find_local_containers
    local_images = get_local_containers()
  File "/home/rajesh/jetson-containers/jetson_containers/container.py", line 351, in get_local_containers
    status = subprocess.run(cmd, stdout=subprocess.PIPE, stderr=subprocess.PIPE,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['sudo', 'docker', 'images', '--format', "'{{json . }}'"]' returned non-zero exit status 1.
-- Error:  return code 1
localuser:root being added to access control list
xauth:  unable to rename authority file /tmp/.docker.xauth, use /tmp/.docker.xauth-n
chmod: changing permissions of '/tmp/.docker.xauth': Operation not permitted
+ sudo docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /var/run/dbus:/var/run/dbus --volume /var/run/avahi-daemon/socket:/var/run/avahi-daemon/socket --volume /home/rajesh/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix -v /tmp/.docker.xauth:/tmp/.docker.xauth -e XAUTHORITY=/tmp/.docker.xauth --device /dev/video0 --device /dev/video1 -e SSL_KEY=/data/key.pem -e SSL_CERT=/data/cert.pem -v /home/rajesh/jetson-containers/packages/llm/local_llm:/opt/local_llm/local_llm python3 -m local_llm.agents.video_query --api=mlc --verbose --model liuhaotian/llava-v1.5-7b --max-new-tokens 32 --video-input /dev/video0 --video-output display://0 --prompt 'Describe the image concisely.'
docker: unknown server OS: .
See 'docker run --help'.

` ` ` 

@dusty-nv - any help?

rajeshroy402, 2024-02-25:
Tried removing entire docker and then followed your guide to reinstall it. 

[image]

But when I do `sudo systemctl restart docker.service` it fails

[image]

rajeshroy402, 2024-02-25:
Logs from `journalctl -eu docker` 

```
Feb 26 00:13:57 jetson-agx systemd[1]: Failed to start Docker Application Container Engine.
Feb 26 00:16:43 jetson-agx systemd[1]: Starting Docker Application Container Engine...
Feb 26 00:16:43 jetson-agx dockerd[5339]: unable to configure the Docker daemon with file /etc/docker/da>
Feb 26 00:16:43 jetson-agx systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 00:16:43 jetson-agx systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 26 00:16:43 jetson-agx systemd[1]: Failed to start Docker Application Container Engine.
Feb 26 00:16:45 jetson-agx systemd[1]: docker.service: Scheduled restart job, restart counter is at 1.
Feb 26 00:16:45 jetson-agx systemd[1]: Stopped Docker Application Container Engine.
Feb 26 00:16:45 jetson-agx systemd[1]: Starting Docker Application Container Engine...
Feb 26 00:16:45 jetson-agx dockerd[5352]: unable to configure the Docker daemon with file /etc/docker/da>
Feb 26 00:16:45 jetson-agx systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 00:16:45 jetson-agx systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 26 00:16:45 jetson-agx systemd[1]: Failed to start Docker Application Container Engine.
Feb 26 00:16:47 jetson-agx systemd[1]: docker.service: Scheduled restart job, restart counter is at 2.
Feb 26 00:16:47 jetson-agx systemd[1]: Stopped Docker Application Container Engine.
Feb 26 00:16:47 jetson-agx systemd[1]: Starting Docker Application Container Engine...
Feb 26 00:16:47 jetson-agx dockerd[5365]: unable to configure the Docker daemon with file /etc/docker/da>
Feb 26 00:16:47 jetson-agx systemd[1]: docker.service: Main process exited, code=exited, status=1/FAILURE
Feb 26 00:16:47 jetson-agx systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 26 00:16:47 jetson-agx systemd[1]: Failed to start Docker Application Container Engine.
Feb 26 00:16:49 jetson-agx systemd[1]: docker.service: Scheduled restart job, restart counter is at 3.
Feb 26 00:16:49 jetson-agx systemd[1]: Stopped Docker Application Container Engine.
Feb 26 00:16:49 jetson-agx systemd[1]: docker.service: Start request repeated too quickly.
Feb 26 00:16:49 jetson-agx systemd[1]: docker.service: Failed with result 'exit-code'.
Feb 26 00:16:49 jetson-agx systemd[1]: Failed to start Docker Application Container Engine.
```

rajeshroy402, 2024-02-25:
OK, so I have this issue resolved. 

I could see that the docker wasn't able to run and from there I followed the error message `[Cannot connect to the Docker daemon at unix:/var/run/docker.sock. Is the docker daemon running?](https://stackoverflow.com/questions/44678725/cannot-connect-to-the-docker-daemon-at-unix-var-run-docker-sock-is-the-docker)`

I reinstalled the docker and this time noticed a new error - 

`failed to start daemon: Error initializing network controller: error obtaining controller instance: unable to add return rule in DOCKER-ISOLATION-STAGE-1 chain.....`

Found out that this has something to do with iptables. After this, I ran the below mentioned two commands - 

```
sudo update-alternatives --set iptables /usr/sbin/iptables-legacy
sudo update-alternatives --set ip6tables /usr/sbin/ip6tables-legacy

```

And then restarted my system. This resolved the issue with my docker. 

I don't know what really caused this problem but surely if someone else comes through this, we have a solution here! 

Thanks

closed 2024-02-25 by rajeshroy402

dusty-nv, 2024-02-25:
@rajeshroy402 I think it's the same issue as https://forums.developer.nvidia.com/t/docker-gives-error-after-upgrading-ubuntu/283563

rajeshroy402, 2024-02-26:
> @rajeshroy402 I think it's the same issue as https://forums.developer.nvidia.com/t/docker-gives-error-after-upgrading-ubuntu/283563

Yes, it's the same issue. 

Thanks, @dusty-nv