dusty-nv/jetson-containers #401: l4t-pytorch with Python 3.9?

sei-jgwohlbier, 2024-02-26:
Related issue:
https://github.com/dusty-nv/jetson-containers/issues/127

I would like to use the python profiling described here that requires 3.9. 
https://docs.nvidia.com/nsight-systems/UserGuide/index.html#python-profiling

I think I found where to update the python version (below). Pointers on rebuilding the required wheels would be much appreciated.

```
diff --git a/packages/python/Dockerfile b/packages/python/Dockerfile
index 2e55dd0..cce3293 100644
--- a/packages/python/Dockerfile
+++ b/packages/python/Dockerfile
@@ -16,7 +16,16 @@ RUN apt-get update && \
                  python3-pip \
     && rm -rf /var/lib/apt/lists/* \
     && apt-get clean
-    
+
+RUN apt-get update && \
+    apt install -y software-properties-common && \
+    add-apt-repository ppa:deadsnakes/ppa && \
+    apt install -y \
+    python3.9 \
+    libpython3.9-dev && \
+    update-alternatives --set python /usr/bin/python3.9
+
 RUN which pip3 && pip3 --version && \
     python3 -m pip install --upgrade pip
```

dusty-nv, 2024-02-27:
Hi @sei-jgwohlbier, what version of JetPack are you on?  if it's JetPack 6.0, you could try `./build.sh pytorch:2.1-builder` .  Otherwise you'll need to make a new config entry for pytorch here:

https://github.com/dusty-nv/jetson-containers/blob/eec7768bc348d8f1e525d8cd322acba865288f5f/packages/pytorch/config.py#L90

sei-jgwohlbier, 2024-02-27:
I'm on 5.1.2. So if I add
`pytorch_build('2.1', suffix='mybuilder', requires='==35.*'),`
combined with the edits to python version in the first comment, and do
`./build.sh pytorch:2.1-mybuilder`
then I should get all new wheels built with the correct stack?

closed 2025-02-21 by johnnynunez