dusty-nv/jetson-containers #403: L4T 36.2.0 (Jetpack 6 DP) - tritonserver doesn't build

Mark-Firestorm, 2024-02-26:
Tritonserver requires libssl1.1 in the current dockerfile, but Ubuntu 22.04 (in Jetpack 6) has moved to libssl3.  I was able to get it to build successfully by adding to the config.py:

```
if L4T_VERSION >= Version('36.2.0'): # JetPack 6.0 DP
    # https://github.com/triton-inference-server/server/releases/tag/v2.42.0
    TRITON_URL = 'https://github.com/triton-inference-server/server/releases/download/v2.42.0/tritonserver2.42.0-igpu.tgz'
    TRITON_TAR = 'tritonserver2.42.0-igpu.tgz'
    TRITON_CLIENTS = 'tritonserver/clients'  # in 2.35, clients/ dir moved under tritonserver/
```

And changing the Dockerfile to install libssl3 on line 30 instead of libssl1.1.  Likely need to create a script to handle which libssl version needs to be installed.

Mark-Firestorm, 2024-02-26:
Alternative to a script to install the packages would be adding another argument to the dockerfile/config.py to store the libssl package name, then swapping it out for the argument value on line 30 of the Dockerfile

dusty-nv, 2024-02-27:
Thanks @Mark-Firestorm, fixed this in https://github.com/dusty-nv/jetson-containers/commit/bd9667998eb28d010891867b82e88ea54f139dbe, and pushed the [`dustynv/tritonserver:r36.2.0`](https://hub.docker.com/repository/docker/dustynv/tritonserver/tags?page=1&ordering=last_updated) container image.

Mark-Firestorm, 2024-02-27:
Thanks @dusty-nv!  I've added another issue for [Deepstream JP6 compatibility](https://github.com/dusty-nv/jetson-containers/issues/405#issue-2155595703) as well which depends on this fix!

closed 2024-02-27 by Mark-Firestorm