dusty-nv/jetson-containers #405: L4T 36.2.0 (Jetpack 6 DP) - deepstream doesn't build

Mark-Firestorm, 2024-02-27:
Deepstream depends on tritonserver which is now updated for JP6.  I was able to get Deepstream to build by adding the following to the config.py:
```
if L4T_VERSION >= Version('36.2.0'): # JetPack 6.0 DP
    DEEPSTREAM_URL = 'https://api.ngc.nvidia.com/v2/resources/org/nvidia/deepstream/6.4/files?redirect=true&path=deepstream_sdk_v6.4.0_jetson.tbz2'
    DEEPSTREAM_TAR = 'deepstream_sdk_v6.4.0_jetson.tbz2'
    PYDS_VERSION = 'v1.1.10'
elif ...
```

The link I found doesn't match the nvidia.box.com links used for the other deepstream downloads, so that might need to be addressed to make things consistent.

I also ran into an issue with building librdkafka, which I resolved by adding a section to the Dockerfile to install python-is-python3 before cloning librdkafka, not sure if this is the right solution but this likely stems from the changing python versions between Ubuntu 20 and 22.  

```
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		python-is-python3 \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean
```

Additionally I had to add tensorrt as a dependency for the deepstream container to resolve an issue with libnvinfer.so.8 not being found.  Not sure if this is the correct solution to building a deepstream container with a minimal environment but it worked for me.

dusty-nv, 2024-02-27:
Thanks again @Mark-Firestorm, fixed this in https://github.com/dusty-nv/jetson-containers/commit/b6c788c2bf1564e93a2159e8710edbaf72df11cd, and rebuilt the DeepStream containers 👍

Mark-Firestorm, 2024-02-27:
Thanks @dusty-nv!

closed 2024-02-27 by Mark-Firestorm