dusty-nv/jetson-containers #409: Deepstream setup adjustments

Mark-Firestorm, 2024-03-04:
When the python bindings are cloned they appear to be cloned in the wrong folder for being able to run the examples.  I'm not sure of the exact directory that the repo is supposed to be cloned in but it is one level deeper than it is currently.  

Proposed fix for this would be on [line 65](https://github.com/dusty-nv/jetson-containers/blob/8c013f71b1b6211260d5b685cd3ca4e7ccc875d0/packages/deepstream/Dockerfile#L65) of the Deepstream dockerfile, changing it to the following:

```
RUN cd /opt/nvidia/deepstream/deepstream && \
```

Evidence for this being in the wrong place can be seen in the [config for deepstream_test1_usbcam](https://github.com/NVIDIA-AI-IOT/deepstream_python_apps/blob/36fdc7ef216d6c4255c6c5931f4adea6f1a1ef0a/apps/deepstream-test1-usbcam/dstest1_pgie_config.txt) where in the Deepstream folder structure, the sources folder being referenced for the model to be used is one level off in the current Docker environment.

dusty-nv, 2024-03-05:
@Mark-Firestorm if you know this fix and could start submitting PRs for these that would be great, I will accept them.  Sorry been really busy, and don't have the time at the moment.

Mark-Firestorm, 2024-03-05:
@dusty-nv I'll put together a fork and a PR now, thanks!

Mark-Firestorm, 2024-03-05:
PR submitted

dusty-nv, 2024-03-05:
Thanks Mark!  I merged your PR https://github.com/dusty-nv/jetson-containers/pull/411 into `dev`.  From there I will merge it into master 👍

Mark-Firestorm, 2024-03-05:
Sure thing, thanks for the quick turn around!

closed 2024-03-05 by Mark-Firestorm