dusty-nv/jetson-containers #410: Pytorch Conv2D produces NaN values on CPU using dustynv:pytorch image.

jneeven, 2024-03-05:
Using [`dustynv/torchvision:r35.2.1`](https://hub.docker.com/layers/dustynv/torchvision/r35.2.1/images/sha256-77924cb1a65245a69ab5a12a2c26829d258c78d3da56eb182c1a618eb5a48fd8?context=explore) on the Orin, I run into NaN issues when performing convolutions on the CPU.

Take the following code:
```python
import torch
import torch.nn.functional as F

print("CPU result (no bias):")
x = F.conv2d(torch.ones(1,3,160,240).cpu(), torch.ones(1,3,3,3).cpu())
print(x)

print("CUDA result (no bias):")
cuda_x = F.conv2d(torch.ones(1,3,160,240).cuda(), torch.ones(1,3,3,3).cuda())
print(cuda_x)

print("CPU result (zero bias):")
x = F.conv2d(torch.ones(1,3,160,240).cpu(), torch.ones(1,3,3,3).cpu(), bias=torch.zeros(1))
print(x)
```

Output:
```
CPU result (no bias):
tensor([[[[27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          ...,
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., nan],
          [nan, 27., 27.,  ..., 27., 27., 27.]]]])
CUDA result (no bias):
tensor([[[[27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          ...,
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.]]]], device='cuda:0')
CPU result (zero bias):
tensor([[[[27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          ...,
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.],
          [27., 27., 27.,  ..., 27., 27., 27.]]]])
```

For some reason, pytorch compiled for the Orin produces NaN values on convolution, but only on the CPU. I have tried  pytorch `2.1.0`  (using the wheel listed [here](https://github.com/dusty-nv/jetson-containers/blob/master/packages/pytorch/config.py#L76C83-L76C210)) and have tried building both containers (2.0.0 and 2.1.0) manually, but it makes no difference...

I suspect the issue is on the Pytorch side, but it might be an issue with the image itself?

Also see https://github.com/pytorch/pytorch/issues/59439#issuecomment-1978385310

dusty-nv, 2024-03-05:
@jneeven there have been aarch64 CPU bugs in pytorch in the past, I would suspect it's an issue in the pytorch source code.  The containers just install the wheels.  I would try installing the CPU-only version of pytorch and see (just `pip3 install torch`)

jneeven, 2024-03-06:
> @jneeven there have been aarch64 CPU bugs in pytorch in the past, I would suspect it's an issue in the pytorch source code. The containers just install the wheels. I would try installing the CPU-only version of pytorch and see (just `pip3 install torch`)

Ah I see, I thought building the containers would build pytorch from scratch as well... In that case let's hope that the issue can be resolved on the pytorch side. Thanks for the quick reply!

closed 2025-02-21 by johnnynunez