dusty-nv/jetson-containers #447: Mamba (State Spaces Models)

johnnynunez, 2024-03-21:
Hi! 
this is only a draft and summary of all papers and implementations of mamba.
Original paper:
(arXiv 2024.01) Vision Mamba: Efficient Visual Representation Learning with Bidirectional State Space Model, [[Paper]](https://arxiv.org/pdf/2401.09417.pdf), [[Code]](https://github.com/bowang-lab/U-Mamba)
(arXiv 2024.01) VMamba: Visual State Space Model, [[Paper]](https://arxiv.org/pdf/2401.10166.pdf), [[Code]](https://github.com/MzeroMiko/VMamba)

(arXiv 2024.03) LocalMamba: Visual State Space Model with Windowed Selective Scan, [[Paper]](https://arxiv.org/pdf/2403.09338.pdf), [[Code]](https://github.com/hunto/LocalMamba)
(arXiv 2024.03) EfficientVMamba: Atrous Selective Scan for Light Weight Visual Mamba, [[Paper]](https://arxiv.org/pdf/2403.09977.pdf), [[Code]](https://github.com/TerryPei/EfficientVMamba)

(arXiv 2024.03) DenseMamba: State Space Models with Dense Hidden Connection for Efficient Large Language Models, [[Paper]](https://arxiv.org/pdf/2403.00818.pdf),[[Code]](https://github.com/WailordHe/DenseSSM)

(arXiv 2024.03) ZigMa: Zigzag Mamba Diffusion Model, [[Paper]](https://arxiv.org/pdf/2403.13802.pdf),[[Code]](https://github.com/CompVis/zigma)

dusty-nv, 2024-03-23:
@johnnynunez that's great! - can you start scoping what an optimized container might look like for supporting Mamba?   is it already supported by other popular LLM API's, or does it have its own?

johnnynunez, 2024-03-23:
> @johnnynunez that's great! - can you start scoping what an optimized container might look like for supporting Mamba? is it already supported by other popular LLM API's, or does it have its own?

it's supported in llama.cpp: https://github.com/ggerganov/llama.cpp/pull/5328

dusty-nv, 2024-03-24:
Hmm that PR says it's CPU-only, and scanning the comments I didn't see mention of GPU support being added.  If you know of a fast CUDA-accelerated implementation to add to jetson-containers, that would be cool.  I don't think it's in MLC...or exllama neither.

What do you think of RKVW also?  That is in MLC, and it's another recurrent one.  Other ppl asked about that one in the past.  It would be cool to do a page on Jetson AI Lab for these recurrent models.

it's supported in llama.cpp: ggerganov/llama.cpp#5328<https://github.com/ggerganov/llama.cpp/pull/5328>

johnnynunez, 2024-04-03:
https://huggingface.co/mlabonne/Jambalpaca-v0.1

johnnynunez, 2024-09-05:
> Hmm that PR says it's CPU-only, and scanning the comments I didn't see mention of GPU support being added. If you know of a fast CUDA-accelerated implementation to add to jetson-containers, that [...]

I've checked the last versions of llama.cpp can run mamba-ssm!! 
https://github.com/ggerganov/llama.cpp/pull/5328

closed 2024-09-14 by johnnynunez