dusty-nv/jetson-containers #448: Packages need to be smarter about building their builders

rgobbel, 2024-03-22:
I've been trying to build a version of local_llm that can load a Mixtral model (a separate issue), and I've noticed that a few of the subpackages fail with a"repository name must be lowercase" error, which really just means that there was a call with `BUILD_IMAGE` set to `None`. This happens when there is a `Dockerfile.builder`, but `find_container` returns `None`. For the cases I've tried, the problem disappears if I first build the builder by hand, so it looks like the general fix would be to always build the builder, if a package has one, but the container is not found.

dusty-nv, 2024-03-22:
Sorry Randy, yes, what builders are missing?

I plan to refactor the whole builder thing and instead have the builders push the wheels to a custom pip pyindex server that I'm gonna run, which should alleviate this issue wholesale.  Plus then ppl will just be able to install my wheels from this custom pyindex without needing to use my containers at all.

rgobbel, 2024-03-22:
I'm not sure of the complete set, but `cuda-pytorch` and `mlc` were definitely failing. There are only a few packages in the entire collection that have these: `cuda-python`, `mlc`, `onnxruntime`, `pytorch`, and `faiss`. I see that you've made a more recent mlc image that knows about Mixtral models, and that's what I'm really interested in, so I'm not going to go through the whole list, but there it is.

dusty-nv, 2024-03-22:
Ok thanks, what is your L4T version?

rgobbel, 2024-03-22:
I'm running JP 6 (36.2.0)