dusty-nv/jetson-containers #450: RIVA doesn't seem to work with DP6 on Orin AGX

jasonthenderson, 2024-03-24:
Poking around I see that Riva says it is only supported up to 5.1, but there are examples in these containers using it and these containers all work with DP6 so I've been trying to no avail, including doing a full reflash and reinstall.  It would be helpful for others if it were clearly noted that it doesn't work with 6 DP so they don't spend time trying to get it to work.

This btw is the error I get....
/opt/riva/bin/start-riva: line 10: curl: command not found
/opt/riva/bin/start-riva: line 11: [: -ne: unary operator expected
  > Triton server is ready...
W0324 18:13:59.715973 19 pinned_memory_manager.cc:236] Unable to allocate pinned system memory, pinned memory pool will not be available: CUDA driver version is insufficient for CUDA runtime version
I0324 18:13:59.716351 19 cuda_memory_manager.cc:115] CUDA memory pool disabled
I0324 18:13:59.724976 19 model_lifecycle.cc:459] loading: conformer-en-US-asr-streaming-ctc-decoder-cpu-streaming:1
I0324 18:13:59.725089 19 model_lifecycle.cc:459] loading: conformer-en-US-asr-streaming-endpointing-streaming:1
I0324 18:13:59.725161 19 model_lifecycle.cc:459] loading: conformer-en-US-asr-streaming-feature-extractor-streaming:1
I0324 18:13:59.725299 19 model_lifecycle.cc:459] loading: intent_slot_detokenizer:1I0324 18:13:59.725239    21 riva_server.cc:126] Using Insecure Server Credentials

I0324 18:13:59.725618 19 model_lifecycle.cc:459] loading: intent_slot_label_tokens_misty:1
I0324 18:13:59.726326 19 model_lifecycle.cc:459] loading: intent_slot_tokenizer-en-US-misty:1
I0324 18:13:59.726721 19 model_lifecycle.cc:459] loading: riva-onnx-fastpitch_encoder-English-US:1
E0324 18:13:59.729030    21 model_registry.cc:286] error: unable to get server status: failed to connect to all addresses; last error: UNKNOWN: ipv4:127.0.0.1:8001: Failed to connect to remote host: Connection refused
I0324 18:13:59.732746 19 model_lifecycle.cc:459] loading: riva-punctuation-en-US:1
I0324 18:13:59.734263 19 model_lifecycle.cc:459] loading: riva-trt-conformer-en-US-asr-streaming-am-streaming:1
I0324 18:13:59.734370 19 model_lifecycle.cc:459] loading: riva-trt-hifigan-English-US:1
I0324 18:13:59.734503 19 model_lifecycle.cc:459] loading: riva-trt-riva-punctuation-en-US-nn-bert-base-uncased:1
I0324 18:13:59.734635 19 model_lifecycle.cc:459] loading: riva-trt-riva_intent_misty-nn-bert-base-uncased:1
I0324 18:13:59.734767 19 model_lifecycle.cc:459] loading: spectrogram_chunker-English-US:1
I0324 18:13:59.734963 19 model_lifecycle.cc:459] loading: tts_postprocessor-English-US:1
I0324 18:13:59.735173 19 model_lifecycle.cc:459] loading: tts_preprocessor-English-US:1
I0324 18:14:00.389257 19 endpointing_library.cc:20] TRITONBACKEND_ModelInitialize: conformer-en-US-asr-streaming-endpointing-streaming (version 1)
E0324 18:14:00.389530 19 model_lifecycle.cc:596] failed to load 'conformer-en-US-asr-streaming-feature-extractor-streaming' version 1: Invalid argument: instance group conformer-en-US-asr-streaming-feature-extractor-streaming_0 of model conformer-en-US-asr-streaming-feature-extractor-streaming has kind KIND_GPU but no GPUs are available
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0324 18:14:00.390765    24 parameter_parser.cc:146] Parameter 'chunk_size' set but unused.
W0324 18:14:00.390797    24 parameter_parser.cc:146] Parameter 'ms_per_timestep' set but unused.
W0324 18:14:00.390801    24 parameter_parser.cc:146] Parameter 'residue_blanks_at_end' set but unused.
W0324 18:14:00.390805    24 parameter_parser.cc:146] Parameter 'residue_blanks_at_start' set but unused.
W0324 18:14:00.390918    24 parameter_parser.cc:146] Parameter 'start_history' set but unused.
W0324 18:14:00.390960    24 parameter_parser.cc:146] Parameter 'start_th' set but unused.
W0324 18:14:00.390986    24 parameter_parser.cc:146] Parameter 'stop_history' set but unused.
W0324 18:14:00.391013    24 parameter_parser.cc:146] Parameter 'stop_th' set but unused.
W0324 18:14:00.391048    24 parameter_parser.cc:146] Parameter 'streaming' set but unused.
W0324 18:14:00.391079    24 parameter_parser.cc:146] Parameter 'use_subword' set but unused.
W0324 18:14:00.391106    24 parameter_parser.cc:146] Parameter 'vocab_file' set but unused.
I0324 18:14:00.392192 19 backend_model.cc:303] model configuration:
{
    "name": "conformer-en-US-asr-streaming-endpointing-streaming",
    "platform": "",
    "backend": "riva_asr_endpointing",
    "version_policy": {
        "latest": {
            "num_versions": 1
        }
    },
    "max_batch_size": 1,
    "input": [
        {
            "name": "CLASS_LOGITS",
            "data_type": "TYPE_FP32",
            "format": "FORMAT_NONE",
            "dims": [
                -1,
                257
            ],
            "is_shape_tensor": false,
            "allow_ragged_batch": false,
            "optional": false
        }
    ],
    "output": [
        {
            "name": "SEGMENTS_START_END",
            "data_type": "TYPE_INT32",
            "dims": [
                -1,
[... 1233 lines cut ...]
I0324 18:14:04.008310 19 pipeline_library.cc:31] TRITONBACKEND_ModelInstanceFinalize: delete instance state
cudaError_t 35 : "CUDA driver version is insufficient for CUDA runtime version" returned from 'cudaHostUnregister(pinned_host_punct_buffer_.data())' in fileriva/nlp/pipeline/punctuator/punctuator.cc line 167'
cudaError_t 35 : "CUDA driver version is insufficient for CUDA runtime version" returned from 'cudaHostUnregister(pinned_host_capit_buffer_.data())' in fileriva/nlp/pipeline/punctuator/punctuator.cc line 168'
I0324 18:14:04.008627 19 sequence_label_cbe.cc:142] TRITONBACKEND_ModelInstanceFinalize: delete instance state
I0324 18:14:04.008697 19 sequence_label_cbe.cc:138] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.008732 19 ctc-decoder-library.cc:27] TRITONBACKEND_ModelInstanceFinalize: delete instance state
I0324 18:14:04.008832 19 detokenizer_cbe.cc:150] TRITONBACKEND_ModelInstanceFinalize: delete instance state
I0324 18:14:04.008845 19 tokenizer_library.cc:27] TRITONBACKEND_ModelInstanceFinalize: delete instance state
I0324 18:14:04.008906 19 detokenizer_cbe.cc:146] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.008666 19 endpointing_library.cc:28] TRITONBACKEND_ModelInstanceFinalize: delete instance state
I0324 18:14:04.009107 19 model_lifecycle.cc:578] successfully unloaded 'intent_slot_detokenizer' version 1
I0324 18:14:04.013533 19 pipeline_library.cc:27] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.014170 19 endpointing_library.cc:23] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.014555 19 tokenizer_library.cc:22] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.008742 19 model_lifecycle.cc:578] successfully unloaded 'intent_slot_label_tokens_misty' version 1
I0324 18:14:04.014816 19 model_lifecycle.cc:578] successfully unloaded 'conformer-en-US-asr-streaming-endpointing-streaming' version 1
I0324 18:14:04.015449 19 model_lifecycle.cc:578] successfully unloaded 'riva-punctuation-en-US' version 1
I0324 18:14:04.022604 19 model_lifecycle.cc:578] successfully unloaded 'intent_slot_tokenizer-en-US-misty' version 1
I0324 18:14:04.945356 19 ctc-decoder-library.cc:24] TRITONBACKEND_ModelFinalize: delete model state
I0324 18:14:04.946123 19 model_lifecycle.cc:578] successfully unloaded 'conformer-en-US-asr-streaming-ctc-decoder-cpu-streaming' version 1
I0324 18:14:05.006539 19 server.cc:302] Timeout 29: Found 0 live models and 0 in-flight non-inference requests
error: creating server: Internal - failed to load all models
E0324 18:14:09.729276    21 model_registry.cc:286] error: unable to get server status: failed to connect to all addresses; last error: UNKNOWN: ipv4:127.0.0.1:8001: Failed to connect to remote host: Connection refused
One of the processes has exited unexpectedly. Stopping container.
W0324 18:14:09.747316    21 riva_server.cc:196] Signal: 15

dusty-nv, 2024-03-24:
Sorry yes the riva-client is fine for JP6, but the riva-embedded server containers from NGC are not out yet for JP6 (should be soon).  So on JP6 you would run the riva server somewhere else or on JP5 instance.  Or use whisper/XTTS for now

jasonthenderson, 2024-03-24:
Thank you for confirming @dusty-nv !

ai-and-i, 2024-03-28:
I got success running riva on JP6 by upgrading ubuntu inside the riva-speech image. Something like this:

```
docker run --name riva_jp6 -ti nvcr.io/nvidia/riva/riva-speech:2.14.0-l4t-aarch64 bash

# inside the docker container
apt-get update && apt-get upgrade -y
apt-get install -y ubuntu-release-upgrader-core
do-release-upgrade  # When prompted, choose to NOT clean obsolete packages
apt-get install -y curl
exit

# outside of the container
docker container commit riva_jp6 nvcr.io/nvidia/riva/riva-speech:2.14.0-l4t-aarch64
```

UserName-wang, 2024-03-28:
@ai-and-i , I tried and it's working now ! Thank you!

dusty-nv, 2024-03-28:
Hi guys, Riva release 2.15 is out now on NGC for JP6, however there is a known issue with the streaming TTS.  That is interesting you got 2.14 to work!

johnnynunez, 2024-03-28:
@dusty-nv now it's not possible to reflash agx with jetpack 6 with the latest version of jp6 and sdk manager

jasonthenderson, 2024-03-29:
Thanks @dusty-nv  and @ai-and-i  - I tried upgrading my own container per @ai-and-i s instructions however it was a 2.12 container.  It still does 'trying in 10 seconds...' but there are no error logs now when I use the docker command it provides at the end of the script.

So then I decided to download 2.15 and again ran into the same error where it displays the 'retrying in 10 seconds...' then when it errors out gives me no error logs when I run the docker riva-speech logs command....

Any ideas on where to look for what is going wrong - very odd to me that both  give that same behavior now.

jasonthenderson, 2024-03-31:
Ok a reboot of the Orin AGX got it so the Riva container is now working.  It seems to be working for both 2.12 and 2.15 though I've now got a strange problem where TTS seems to be using the wrong framerate...my usb speaker says that it is 16khz and in fact I'm unable to set any other framerate in talk.py...but it sounds like Alvin and the Chipmunks...so seems like for some reason the speed isn't getting set right.  This used to work in JP5.1 so I'm not sure if there is some additional setting I need to do...

@dusty-nv  - I saw you post that you were setting up xtts - do you have any instructions posted on how to use that for streaming out to a speaker?

closed 2025-02-21 by johnnynunez