dusty-nv/jetson-containers #457: About Temperature and Noise of Jetson Agx Orin 64G Processor

lq0104, 2024-04-02:
This is a great project, and I would like to ask a question: what is the temperature of the Jetson Agx Orin 64G processor when running llamaspeak or other large language models, and is the noise level of the machine when running the program

lq0104, 2024-04-02:
Because I think the level of noise is an important factor in human-computer interaction

dusty-nv, 2024-04-02:
It does have a fan but I've rarely heard it, if ever.  The current temp running vision/language model shows 41C...it's not exactly like a PC, there are fanless/sealed embedded systems built for Jetson

lq0104, 2024-04-02:
The sound is low, and the temperature is also very low. It's really cool! May I ask if your Jetson was assembled using a module by yourself or is it an official shelf product? Thank you very much for your answer

dusty-nv, 2024-04-02:
It's just the devkit

lq0104, 2024-04-02:
That's amazing! Thank you~

closed 2024-04-02 by lq0104