dusty-nv/jetson-containers #464: is:issue Error while finding module specification for 'local_llm.test.asr' (ModuleNotFoundError: No module named 'local_llm.test')

cfregly, 2024-04-04:
local_llm.test module does not exist in latest docker image https://hub.docker.com/layers/dustynv/local_llm/r36.2.0/images/sha256-9533873e7ece872d154c8302036d896adb7b3b79e91337fdd8cedcdd67c20e89?context=explore

any idea what happened to that?

affects commands like this:

```
./run.sh dustynv/local_llm:r36.2.0 \
  python3 -m local_llm.test.asr --list-audio-devices
```

cfregly, 2024-04-04:
```
root@jetson:/# ls -al /opt/local_llm/local_llm/
```

```
total 84
drwxr-xr-x 1 root root  4096 Mar 15 14:52 .
drwxr-xr-x 1 root root  4096 Mar 14 00:01 ..
-rw-rw-r-- 1 root root  4979 Jan 24 03:56 agent.py
drwxr-xr-x 3 root root  4096 Mar 15 14:52 agents
drwxr-xr-x 3 root root  4096 Mar 15 14:52 chat
-rw-rw-r-- 1 root root  1750 Mar  1 06:25 completion.py
-rw-rw-r-- 1 root root   278 Mar  1 06:13 __init__.py
-rw-rw-r-- 1 root root 10120 Mar 15 07:45 local_llm.py
-rw-rw-r-- 1 root root   358 Mar  1 16:49 __main__.py
drwxr-xr-x 3 root root  4096 Mar 15 14:52 models
-rw-rw-r-- 1 root root  9726 Mar 15 07:36 plugin.py
drwxr-xr-x 4 root root  4096 Mar 15 14:52 plugins
-rw-rw-r-- 1 root root    51 Jan 25 23:57 requirements.txt
drwxr-xr-x 3 root root  4096 Mar 15 14:52 utils
drwxr-xr-x 3 root root  4096 Mar 15 14:52 vision
drwxr-xr-x 5 root root  4096 Mar 15 14:52 web
```

cfregly, 2024-04-04:
this worked previously... until I did an update on this container today.

i manually copied from here:  https://github.com/dusty-nv/jetson-containers/tree/7bf8827/packages/llm/local_llm

and now it works.

does this image need to be rebuilt?  or should I switch to a different image?  concerned that other stuff might be out of sync.

dusty-nv, 2024-04-04:
Ahh yes sorry, I need to push the latest/final after I do some more testing on it (I changed a bunch of stuff in the CUDA/Python base containers)

For now you can revert commit https://github.com/dusty-nv/jetson-containers/commit/1a7fb07dd6183be744784a1c418799c1e0796ca3 and add this back into run.sh (or do the mount yourself, like it sounds like you are doing now)

```
--volume $ROOT/packages/llm/local_llm:/opt/local_llm/local_llm
```

dusty-nv, 2024-04-05:
@cfregly FYI I reverted this in the `dev` branch in commit https://github.com/dusty-nv/jetson-containers/commit/7164b42aecdf0540c1fddc2adabc322a5c2dd316 until it's properly fixed

closed 2025-02-21 by johnnynunez