dusty-nv/jetson-containers #472: [Import Error] Unable to import tensorrt inside pytorch:2.1-r36.2.0

mzacri, 2024-04-10:
Hi everyone,

I have installed the pytorch:2.1-r36.2.0 image and i'm having troubles importing tensorrt.

**Error output**
>> import tensorrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.10/dist-packages/tensorrt/__init__.py", line 67, in <module>
    from .tensorrt import *
ImportError: /usr/local/cuda/compat/libnvcudla.so: undefined symbol: _ZN5nvdla8IRuntime22registerTaskStatisticsE15NvDlaMemDescRec15NvDlaAccessType

**Outside container env**
_Hardware:_ Jetson Orin 32GB Custom board
_JetPack Release:_ 5.1 (R35.2.1)
_TRT:_ TensorRT 8.5.2.2.1+cuda11.4
_Torch:_ Torch 2.1.0a0+41361538.nv23.6

I have checked that libnvdla drivers are installed outside container:
-rw-r--r-- 1 root root 6978312 janv. 25  2023 /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so
-rw-r--r-- 1 root root  404792 janv. 25  2023 /usr/lib/aarch64-linux-gnu/tegra/libnvdla_runtime.so

Any troubleshooting paths would be welcomed.

Thank you!

dusty-nv, 2024-04-10:
Hi @mzacri, you are trying to run a container image for JetPack 6 (r36.2) on JetPack 5 (r35.2)  - try running one of the r35 images instead

johnnynunez, 2024-04-10:
You also can compile it:
I have scripts, but change your version and python version links
with python 3.11
```bash
export EXT_PATH=~/external
export TRT_OSSPATH=~/external/TensorRT

mkdir -p $EXT_PATH && cd $EXT_PATH
git clone https://github.com/pybind/pybind11.git

wget https://www.python.org/ftp/python/3.11.6/Python-3.11.6.tgz
tar -xvf Python-3.11.6.tgz
mkdir -p $EXT_PATH/python3.11/include
cp -r Python-3.11.6/Include/* $EXT_PATH/python3.11/include

wget https://deb.menhera.org/ubuntu-ports/pool/main/p/python3.11/libpython3.11-dev_3.11.6-3_arm64.deb
ar x libpython3.11-dev*.deb
mkdir debian && tar -xf data.tar.zst -C debian
cp debian/usr/include/aarch64-linux-gnu/python3.11/pyconfig.h python3.11/include/

git clone --branch release/8.6 --recursive https://github.com/NVIDIA/TensorRT.git
cd TensorRT
mkdir -p build && cd build
export TRT_LIBPATH=/usr/lib/aarch64-linux-gnu/
cmake .. -DTRT_LIB_DIR=$TRT_LIBPATH -DTRT_OUT_DIR=`pwd`/out -DTRT_PLATFORM_ID=aarch64 -DCUDA_VERSION=12.2 -DGPU_ARCHS="87"
CC=/usr/bin/gcc make -j$(nproc)

cd ../python
TENSORRT_MODULE=tensorrt PYTHON_MAJOR_VERSION=3 PYTHON_MINOR_VERSION=11 TARGET_ARCHITECTURE=aarch64 TRT_OSSPATH=~/external/TensorRT ./build.sh
pip install ./build/bindings_wheel/dist/tensorrt-*.whl
```

mzacri, 2024-04-11:
Hi @dusty-nv ,

I have tried a JP5 image and it works fine. Thanks !

@johnnynunez , i will try your method also when i'll have some time to dig further :). Thanks.

closed 2025-02-21 by johnnynunez