dusty-nv/jetson-containers #473: Can't build librealsense - missing python 3.6m - R.32.7.4/Ubuntu 20.04

pgluz, 2024-04-10:
Hi. 

First, thanks for awesome job you did.
I've installed jetson-containers and tried to build an image with `jetson-containers build --name=my_name realsense ros:humble-desktop
`

Everything works fine until it starts building `librealsense`:
```
-- Fetching recommended firmwares:
-- disable link time optimization for fw project
-- ... https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-5.15.1.0.bin
-- ... https://librealsense.intel.com/Releases/SR300/FW/SR3XX_FW_Image-3.26.1.0.bin
-- ... https://librealsense.intel.com/Releases/L5xx/FW/L51X_FW_Image-1.5.8.1.bin
-- Configuring done (23.5s)
CMake Error in third-party/rsutils/py/CMakeLists.txt:
  Imported target "pybind11::module" includes non-existent path

    "/usr/include/python3.6m"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in wrappers/python/CMakeLists.txt:
  Imported target "pybind11::module" includes non-existent path

    "/usr/include/python3.6m"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

CMake Error in wrappers/python/CMakeLists.txt:
  Imported target "pybind11::module" includes non-existent path

    "/usr/include/python3.6m"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done (0.7s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
The command '/bin/sh -c git clone --branch ${LIBREALSENSE_VERSION} --depth=1 https://github.com/IntelRealSense/librealsense &&     cd librealsense &&     mkdir build &&     cd build &&     cmake         -DBUILD_EXAMPLES=true 	   -DFORCE_RSUSB_BACKEND=true 	   -DBUILD_WITH_CUDA=true 	   -DCMAKE_BUILD_TYPE=release 	   -DBUILD_PYTHON_BINDINGS=bool:true 	   -DPYTHON_EXECUTABLE=/usr/bin/python3 	   -DPYTHON_INSTALL_DIR=$(python3 -c 'import sys; print(f"/usr/lib/python{sys.version_info.major}.{sys.version_info.minor}/dist-packages")') 	   ../ &&     make -j$(($(nproc)-1)) &&     make install &&     cd ../ &&     cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/ &&     rm -rf librealsense' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/jetson/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/jetson/jetson-containers/jetson_containers/container.py", line 143, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag typeone:l4t-r32.7.4-realsense --file /home/jetson/jetson-containers/packages/realsense/Dockerfile --build-arg BASE_IMAGE=typeone:l4t-r32.7.4-cmake /home/jetson/jetson-containers/packages/realsense 2>&1 | tee /home/jetson/jetson-containers/logs/20240410_155919/build/typeone_l4t-r32.7.4-realsense.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

I use Ubuntu 20.04 with R32 rev. 7.4, kernel 4.9.337-tegra. My python3 version is 3.8.10.

I will be extremely grateful for any suggestions what to fix it. 

I also noticed minor warning during previous steps of building the image:
```
CMake Deprecation Warning at common/fw/CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
```

dusty-nv, 2024-04-10:
@pgluz I haven't built librealsense in a minute and so it may need pinned for that older JetPack, also you are on an odd system setup somehow running Ubuntu 20.04 on JetPack 4 / L4T R32 (which can have impacts in the containers because of how system files are mounted into the containers on JetPack 4)

pgluz, 2024-04-11:
@dusty-nv Unfortunately, JetPack 4.6.4 is the latest version for Jetson Nano. After upgrade JetPack it works pretty well on Ubuntu 20.04 from Qengineering, even BT headers 8265NGW has been installed correctly and CUDA also works.

I downgraded Ubuntu to the official 18.04 with the same JetPack and the installation went further but failed with the errors below:
` copying _skbuild/linux-aarch64-3.6/cmake-install/share/opencv4/haarcascades/haarcascade_upperbody.xml -> _skbuild/linux-aarch64-3.6/cmake-install/cv2/data/haarcascade_upperbody.xml
  Traceback (most recent call last):
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 262, in build_wheel
      metadata_directory)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 231, in build_wheel
      wheel_directory, config_settings)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
      self.run_setup()
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 268, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 541, in <module>
      main()
    File "setup.py", line 314, in main
      cmake_source_dir=cmake_source_dir,
    File "/tmp/pip-build-env-ri4glkye/overlay/lib/python3.6/site-packages/skbuild/setuptools_wrap.py", line 683, in setup
      cmake_install_dir,
    File "setup.py", line 454, in _classify_installed_files_override
      raise Exception("Not found: '%s'" % relpath_re)
  Exception: Not found: 'python/cv2/mat_wrapper/.*\.py'
  Building wheel for opencv-contrib-python (pyproject.toml): finished with status 'error'
  ERROR: Failed building wheel for opencv-contrib-python
Failed to build opencv-contrib-python
ERROR: Failed to build one or more wheels
The command '/bin/sh -c /tmp/opencv/opencv_install_deps.sh &&     pip3 install --no-cache-dir 	 opencv-contrib-python~=${OPENCV_VERSION}     || /tmp/opencv/build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/jetson/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/jetson/jetson-containers/jetson_containers/container.py", line 143, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag typeone:l4t-r32.7.4-opencv --file /home/jetson/jetson-containers/packages/opencv/Dockerfile.pip --build-arg BASE_IMAGE=typeone:l4t-r32.7.4-numpy --build-arg OPENCV_VERSION="4.5.0" --build-arg OPENCV_PYTHON="4.x" --build-arg CUDA_ARCH_BIN="5.3,6.2,7.2" /home/jetson/jetson-containers/packages/opencv 2>&1 | tee /home/jetson/jetson-containers/logs/20240410_221805/build/typeone_l4t-r32.7.4-opencv.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.`

Maybe symlinks to this missing files would be enough? Unfortunately, I can't do that now, but I will in my spare time.
At this moment I've sticked to community-contributed ROS2 docker image. Anyway, thank you for response!

closed 2025-02-21 by johnnynunez