dusty-nv/jetson-containers #478: Container's directory name needs to be unique regardless of path

ms1design, 2024-04-17:
Hi @dusty-nv 👋 

I've created a `wyoming` wrapper for existing `faster-whisper` container called `wyoming-faster-whisper`. Unfortunately, I've encountered an issue where I can't use the same container directory names even if they're in different paths. For instance:

- `faster-whisper` container is located in `packages/audio/faster-whisper`
- the new `wyoming-faster-whisper` container is located in `packages/smart-home/wyoming/faster-whisper`

As a result, `faster-whisper` is detected, but `wyoming-faster-whisper` is not. However, if we change its path to anything other than `.../faster-whisper`, it works as expected. Not sure if this was intended?

Best,
M

dusty-nv, 2024-04-17:
Can you try manually setting the name in the dockerfile yaml header or config.py?   And yes, it gets the initial package name from the directory before parsing the dockerfile and config files

ms1design, 2024-04-17:
Tried to fix that using `config.py`, but setting the `name` or `alias` there does nothing. Only the dir name change helps.

dusty-nv, 2024-04-17:
@ms1design will look at this later, but for now just create it under `wyoming/whisper` .  I think that should be fine anyways as we don't really wanna maintain multiple whisper variants for wyoming anyways

ms1design, 2024-04-17:
That's sounds reasonable, thanks for explanations @dusty-nv 🙏

ms1design, 2025-02-21:
@johnnynunez was this resolved? 🤔

johnnynunez, 2025-02-21:
> [@johnnynunez](https://github.com/johnnynunez) was this resolved? 🤔

I’m cleaning issues for future bot and the use of new template. I reopen it