dusty-nv/jetson-containers #480: Unable to build custom Docker

bryanhughes, 2024-04-18:
Tried to build a custom Docker with the following:
```
$ ./jetson-containers build --name=mimzy opencv:4.9.0 whisper piper-tts nano_llm:main
```
After a long while, it errored out:
```
Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
+ ln -s /usr/bin/llvm-config-14 /usr/bin/llvm-config-17 /usr/bin/llvm-config
ln: target '/usr/bin/llvm-config': Not a directory
The command '/bin/bash -c /tmp/mlc/install.sh || /tmp/mlc/build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/mnt/nvme/git/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/mnt/nvme/git/jetson-containers/jetson_containers/container.py", line 143, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo DOCKER_BUILDKIT=0 docker build --network=host --tag mimzy:r36.2.0-mlc --file /mnt/nvme/git/jetson-containers/packages/llm/mlc/Dockerfile --build-arg BASE_IMAGE=mimzy:r36.2.0-nanodb --build-arg MLC_VERSION="0.1" --build-arg MLC_COMMIT="607dc5a" --build-arg MLC_PATCH="patches/607dc5a.diff" --build-arg TVM_VERSION="0.15" --build-arg LLVM_VERSION="17" /mnt/nvme/git/jetson-containers/packages/llm/mlc 2>&1 | tee /mnt/nvme/git/jetson-containers/logs/20240417_171330/build/mimzy_r36.2.0-mlc.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

The error seems to be:
```
ln -s /usr/bin/llvm-config-14 /usr/bin/llvm-config-17 /usr/bin/llvm-config
```
These files do not exist.

[mimzy_r36.2.0-mlc.txt](https://github.com/dusty-nv/jetson-containers/files/15018647/mimzy_r36.2.0-mlc.txt)

ms1design, 2024-04-18:
Hi @bryanhughes :)

Could you try to remove the `openai-triton` from `onnxruntime` container `Dockerfile` dependencies as shown below:

```
#---
# name: onnxruntime
# group: ml
# config: config.py
# depends: [cuda, cudnn, tensorrt, cmake, python, numpy, onnx]
# test: test.py
# notes: the `onnxruntime-gpu` wheel that's built is saved in the container under `/opt`
#---
```

The mentioned `llvm ` is a dependency of `openai-triton` package

bryanhughes, 2024-04-19:
@ms1design Thanks! Turns out I should have been working off of the `dev` branch for now, which I am and now everything works. Huzza! Except for my new issue which I am filing now.

Cheers!

closed 2024-04-19 by bryanhughes