dusty-nv/jetson-containers #484: Unable to play audio from in the Docker

bryanhughes, 2024-04-19:
I successfully built a custom docker which included me creating a custom package where I needed to install the dependencies for the application that I am writing. After much trials this works great and everything is good.

My problem is that I am mounting my code base into the docker to run and develop my application. It uses pyaudio to record and playback audio from the attached Anker PowerConf S3 mic/speaker. This all works perfectly on the host but doesnt in the docker.

I can record audio and do a speaker-test with no problems, just playing back the audio doesnt work. I get 

```
# arecord -f S16_LE -c 1 -d 10 -r 48000 --device="hw:2,0" /tmp/test-mic.wav
Recording WAVE '/tmp/test-mic.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

# aplay /tmp/test-mic.wav 
aplay: main:831: audio open error: No such file or directory

# ll /tmp/test-mic.wav 
-rw-r--r-- 1 root root 960044 Apr 18 19:57 /tmp/test-mic.wav
```

I found this solution, but it didnt work for me:

https://forums.developer.nvidia.com/t/how-to-access-usb-speaker-from-within-a-docker-container/231737

Here is my custom run script:

```
./jetson-containers run --volume /mnt/nvme/git/mimzy:/opt/mimzy --device /dev/snd --device /dev/bus/usb --volume /etc/timezone:/etc/timezone:ro --volume /etc/localtime:/etc/localtime:ro mimzy:r36.2.0
```

dusty-nv, 2024-04-19:
@bryanhughes `jetson-containers run` (aka run.sh) already mounts the `/dev/snd` and `/dev-bus/usb` audio devices:

https://github.com/dusty-nv/jetson-containers/blob/4c4d7424c3b4d53b916ec2f20ebda222f3fd82b0/run.sh#L65

Strangely I have the same Anker PowerConf usb and it works through pyaudio both for mic and speaker inside docker....
With USB devices, you also need to have them plugged in before you start the container.

Have you listed the audio devices inside container with a script like this?   

https://github.com/dusty-nv/jetson-containers/blob/master/packages/audio/riva-client/list_audio_devices.py

ms1design, 2024-04-19:
Oh how nice, I'm on the same stage as @bryanhughes right now with my Anker S330 and Home Assistant containers :D

bryanhughes, 2024-04-19:
> https://github.com/dusty-nv/jetson-containers/blob/master/packages/audio/riva-client/list_audio_devices.py

@dusty-nv I have which corroborates what `aplay -l` shows 

```
AUDIO DEVICES:

 0: NVIDIA Jetson AGX Orin HDA: HDMI 1 (hw:0,7)        (inputs=0   outputs=8   sampleRate=44100)
 1: NVIDIA Jetson AGX Orin HDA: HDMI 2 (hw:0,8)        (inputs=0   outputs=8   sampleRate=44100)
 2: NVIDIA Jetson AGX Orin HDA: HDMI 3 (hw:0,9)        (inputs=0   outputs=8   sampleRate=44100)
 3: NVIDIA Jetson AGX Orin APE: - (hw:1,0)             (inputs=16  outputs=16  sampleRate=44100)
 4: NVIDIA Jetson AGX Orin APE: - (hw:1,1)             (inputs=16  outputs=16  sampleRate=44100)
 5: NVIDIA Jetson AGX Orin APE: - (hw:1,2)             (inputs=16  outputs=16  sampleRate=44100)
 6: NVIDIA Jetson AGX Orin APE: - (hw:1,3)             (inputs=16  outputs=16  sampleRate=44100)
 7: NVIDIA Jetson AGX Orin APE: - (hw:1,4)             (inputs=16  outputs=16  sampleRate=44100)
 8: NVIDIA Jetson AGX Orin APE: - (hw:1,5)             (inputs=16  outputs=16  sampleRate=44100)
 9: NVIDIA Jetson AGX Orin APE: - (hw:1,6)             (inputs=16  outputs=16  sampleRate=44100)
10: NVIDIA Jetson AGX Orin APE: - (hw:1,7)             (inputs=16  outputs=16  sampleRate=44100)
11: NVIDIA Jetson AGX Orin APE: - (hw:1,8)             (inputs=16  outputs=16  sampleRate=44100)
12: NVIDIA Jetson AGX Orin APE: - (hw:1,9)             (inputs=16  outputs=16  sampleRate=44100)
13: NVIDIA Jetson AGX Orin APE: - (hw:1,10)            (inputs=16  outputs=16  sampleRate=44100)
14: NVIDIA Jetson AGX Orin APE: - (hw:1,11)            (inputs=16  outputs=16  sampleRate=44100)
15: NVIDIA Jetson AGX Orin APE: - (hw:1,12)            (inputs=16  outputs=16  sampleRate=44100)
16: NVIDIA Jetson AGX Orin APE: - (hw:1,13)            (inputs=16  outputs=16  sampleRate=44100)
17: NVIDIA Jetson AGX Orin APE: - (hw:1,14)            (inputs=16  outputs=16  sampleRate=44100)
18: NVIDIA Jetson AGX Orin APE: - (hw:1,15)            (inputs=16  outputs=16  sampleRate=44100)
19: NVIDIA Jetson AGX Orin APE: - (hw:1,16)            (inputs=16  outputs=16  sampleRate=44100)
20: NVIDIA Jetson AGX Orin APE: - (hw:1,17)            (inputs=16  outputs=16  sampleRate=44100)
21: NVIDIA Jetson AGX Orin APE: - (hw:1,18)            (inputs=16  outputs=16  sampleRate=44100)
22: NVIDIA Jetson AGX Orin APE: - (hw:1,19)            (inputs=16  outputs=16  sampleRate=44100)
23: PowerConf S3: USB Audio (hw:2,0)                   (inputs=1   outputs=2   sampleRate=48000)
```
Unfortunately when trying to play an audio I get:
```
# aplay mimzy-1.wav 
aplay: main:831: audio open error: No such file or directory
```
My current approach is to debug by getting `aplay` to work from the command line in the running docker.

bryanhughes, 2024-04-19:
```
# aplay --device "hw:2,0" mimzy-1.wav
Playing WAVE 'mimzy-1.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
aplay: set_params:1358: Channels count non available
```
Googling the message looks like it leads down the path of modifying the `alsa` config, which I did once upon a time and ended up screwing up my system. Just to clarify, `aplay` works fine on the host.

bryanhughes, 2024-04-19:
Also, this works:
```
speaker-test -f S16_LE -c 2 -d 10 -r 48000 --device="hw:2,0"
```

ms1design, 2024-04-19:
This is somewhat related with my issues with `wyoming-assist-microphone` container here: https://github.com/dusty-nv/jetson-containers/pull/481

```
DEBUG:root:Connected to mic service
arecord: main:831: audio open error: No such file or directory
WARNING:root:Mic service disconnected
```

bryanhughes, 2024-04-19:
@ms1design Unfortunately I still get:

```
# aplay mimzy-1-converted.wav 
aplay: main:831: audio open error: No such file or directory
```
When I do:
```
aplay --device="hw:2,0"  mimzy-1-converted.wav 
aplay: main:831: audio open error: Device or resource busy
```
This is because The PowerConf is current the default device on the host set in Settings (I am on the desktop). When I switch the host default output to the HDMI speaker in the monitor, I get the following on the original and converted:
```
# aplay --device="hw:2,0" mimzy-1.wav 
Playing WAVE 'mimzy-1.wav' : Signed 16 bit Little Endian, Rate 22050 Hz, Mono
aplay: set_params:1358: Channels count non available

# aplay --device="hw:2,0" mimzy-1-converted.wav
Playing WAVE 'mimzy-1-converted.wav' : Signed 24 bit Little Endian in 3bytes, Rate 48000 Hz, Stereo
aplay: set_params:1352: Sample format non available
Available formats:
- S16_LE
```
I tried adding `-f S16_LE` but no difference. Stranger and stranger.

bryanhughes, 2024-04-19:
More information trying out `speaker-test`. When the PowerConf is set as the default output on the host through Settings, I can generate the test sound. When I change the default to the HDMI monitor speaker on the host through Settings, and then try to run the speaker test specifying the device:
```
$ speaker-test --device="hw:2,0"

speaker-test 1.2.6

Playback device is hw:2,0
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Channels count (1) not available for playbacks: Invalid argument
Setting of hwparams failed: Invalid argument
``` 
It doesnt work on the host, so at least the host and the docker are failing in the same way when specifying the device.

ms1design, 2024-04-20:
That's interesting, @bryanhughes 🤔 I've finished enabling `CUDA` support for the `wyoming-piper` wrapper for the `piper-tts` container. Now, I can experiment with a setup similar to yours (Orin, Anker S330, containers, etc.). Did you modify your `ALSA` configuration this time?

bryanhughes, 2024-04-20:
@ms1design I did not modify the ALSA configuration. What I am doing is creating a custom container with just nano_llm, whisper, and piper-tts. I discovered another bug when trying to use lamaspeaks with whisper and piper, if throws an error unless it is riva.

I just rebuilt my container, this time I added in `apt install alsa-base`, but no changes. I want to test out if riva needs to be included and maybe that is where the audio magic happens?

bryanhughes, 2024-04-21:
Just an Update: I added `riva-client` to my custom build, to see if maybe there was some magic there, and it is also crapping out when I run the llama speaks demo. I am thinking that maybe my Orin AGX may be the problem.

bryanhughes, 2024-04-21:
So I am deep down the rabbit hole. It is clear this issue is not related to Docker, but to my host system, which I just re-flashed. Here is the updated information I posted in the Jetson Developer Forum:

https://forums.developer.nvidia.com/t/how-to-access-usb-speaker-from-within-a-docker-container/290183/4

dusty-nv, 2024-04-21:
@bryanhughes @ms1design given the problems you were having, just went through this and found aplay doesn't like it when the wav file is mono and output device is stereo, because hw devices won't do format conversion or resampling for you:

https://forums.developer.nvidia.com/t/how-to-access-usb-speaker-from-within-a-docker-container/290183/6?u=dusty_nv

Don't believe it is related to Jetson, or your device needing re-flashed.  The alsa-utils are cryptic to use and if you are developing for Python I would just use PyAudio.

bryanhughes, 2024-04-21:
@dusty-nv Thank you. I will let you know the progress. I know more about `alsa` than I ever wanted to and am just realizing this. There are so many red-herrings doh!

bryanhughes, 2024-04-21:
@dusty-nv Just an update. I decided to take a step back and just see if the llama speaks demo runs, which it sort of does. The TTS part works but the ASR didnt, which I realize is because the Riva server needs to be started first. I decided to switch to the `whisper` ASR but unfortunately looks like the code is hardwired to `riva`:

```
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/opt/NanoLLM/nano_llm/agents/web_chat.py", line 150, in <module>
    agent = WebChat(**vars(args))
  File "/opt/NanoLLM/nano_llm/agents/web_chat.py", line 30, in __init__
    super().__init__(**kwargs)
  File "/opt/NanoLLM/nano_llm/agents/voice_chat.py", line 38, in __init__
    self.asr = AutoASR.from_pretrained(asr=asr, **kwargs) 
  File "/opt/NanoLLM/nano_llm/plugins/audio/auto_asr.py", line 41, in from_pretrained
    raise ValueError(f"ASR model type should be 'riva'")
ValueError: ASR model type should be 'riva'
```
From the documentation, it looks like Riva is built into the container, but not super clear on how to start it, and then start llama speaks.

ms1design, 2024-04-22:
@dusty-nv @bryanhughes Now I'm happy that I spent that weekend resting rather than deep diving into `ALSA` :D 

## Testing compatibility

First thing I did was just test using `wyoming-assist-microphone` container – what output I get on `host` Orin / `docker run` / `docker-compose` when issuing below commands:

- `aplay -L`
- `arecord -L`

### `host` – ✅ Mic/Speaker detected / working as expected

<details><summary><code>aplay -L</code> on <code>host</code></summary>
<p>

```bash
narandill@ubuntu:~$ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
default
    Playback/recording through the PulseAudio sound server
demixer
hw:CARD=HDA,DEV=3
    NVIDIA Jetson AGX Orin HDA, HDMI 0
    Direct hardware device without any conversions
hw:CARD=HDA,DEV=7
    NVIDIA Jetson AGX Orin HDA, HDMI 1
    Direct hardware device without any conversions
hw:CARD=HDA,DEV=8
    NVIDIA Jetson AGX Orin HDA, HDMI 2
    Direct hardware device without any conversions
hw:CARD=HDA,DEV=9
    NVIDIA Jetson AGX Orin HDA, HDMI 3
    Direct hardware device without any conversions
plughw:CARD=HDA,DEV=3
    NVIDIA Jetson AGX Orin HDA, HDMI 0
    Hardware device with all software conversions
plughw:CARD=HDA,DEV=7
    NVIDIA Jetson AGX Orin HDA, HDMI 1
    Hardware device with all software conversions
plughw:CARD=HDA,DEV=8
    NVIDIA Jetson AGX Orin HDA, HDMI 2
    Hardware device with all software conversions
plughw:CARD=HDA,DEV=9
    NVIDIA Jetson AGX Orin HDA, HDMI 3
    Hardware device with all software conversions
hdmi:CARD=HDA,DEV=0
    NVIDIA Jetson AGX Orin HDA, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDA,DEV=1
    NVIDIA Jetson AGX Orin HDA, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDA,DEV=2
    NVIDIA Jetson AGX Orin HDA, HDMI 2
    HDMI Audio Output
dmix:CARD=HDA,DEV=3
    NVIDIA Jetson AGX Orin HDA, HDMI 0
[... 1683 lines cut ...]

### Unfortunately I cannot directly play the recording without transcoding from `S16_LE` to `S24_3LE` sample format using `sox`:
❌ **This fails**:
```bash
narandill@ubuntu:~$ aplay -D hw:2,0 -f S16_LE -r 48000 recording.wav
Playing WAVE 'recording.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1352: Sample format non available
Available formats:
- S24_3LE
narandill@ubuntu:~$
```
✅ **This works**:
```bash
narandill@ubuntu:~$ sox recording.wav -b 24 -c 2 -r 48000 -e signed-integer recording-converted.wav
narandill@ubuntu:~$ aplay -D hw:2,0 -f S24_3LE -r 48000 recording-converted.wav
Playing WAVE 'recording-converted.wav' : Signed 24 bit Little Endian in 3bytes, Rate 48000 Hz, Stereo
narandill@ubuntu:~$
```

### Working commands:
```bash
arecord -D hw:2,0 -c 2 -r 48000 -f S16_LE -d 5 -t wav recording.wav
sox recording.wav -b 24 -c 2 -r 48000 -e signed-integer recording-converted.wav
aplay -D hw:2,0 -f S24_3LE -r 48000 recording-converted.wav
```

dusty-nv, 2024-04-22:
> `jetson-containers` tests – 💩 Mic/Speaker not available

@ms1design I think this is because jetson-containers does not mount devices like `/dev/snd` and `/dev/usb` for tests, it doesn't mount `/dev/video` devices either

> My Anker device does not allow to record in different rate than 48000Hz
> My Anker want's to record only with two channels (-c 2 instead of -c 1).

Same here and I believe expected behavior for the Anker device, based on other sound devices I have that you also need to specify the correct sample rate and number of channels for.

bryanhughes, 2024-04-22:
@dusty-nv and @ms1design - Just wanted to add what I learned this weekend diving down the `alsa` pit of despair. The part that threw me was that the default driver (on my system ID 37) does resampling. Meaning that whatever I played both through `aplay` and my `pyaudio` output worked. Only when reading and writing directly to the card is when there is no resampling for you. As Dusty pointed out in my post to the Dev Forum, the `alsa` was just a giant waste of my time (DOH). 

So for ASR (using whisper) I downsample from 48000 to 16000, and then for TTS (using the Cori voice on Piper) I upsample from 22050 to 48000. Everything is working nicely now.

Thanks again @dusty-nv for you help and patience. I will respond to my post on the dev forum with the final solution.

dusty-nv, 2024-04-22:
If you guys wanna see how I handle resampling from Python, I use torchaudio, which is faster than librosa and can use CUDA:

https://github.com/dusty-nv/NanoLLM/blob/d493ce48e5b0f4cf9fe795c8366a733c46c5d651/nano_llm/plugins/audio/piper_tts.py#L77

(this would be for if/when you control or patch the code in question, as opposed to trying to get another existing 3rd-party project running without modification)

ms1design, 2024-04-22:
> If you guys wanna see how I handle resampling from Python, I use torchaudio, which is faster than librosa and can use CUDA:

I think that there's no need for that - according to the [`assist-micophone` docs](https://github.com/rhasspy/wyoming-satellite/blob/962b2f5dc6ed0d8b05671f1ade092407ed6251cd/docs/tutorial_2mic.md#determine-audio-devices) you can also make recording with different sampling, channels and just one profile:

```bash
arecord -D plughw:CARD=S330,DEV=0 -r 16000 -c 1 -f S16_LE -t wav -d 5 test.wav
aplay -D plughw:CARD=S330,DEV=0 test.wav
``` 

The above works when device is defined following the docs: `-D plughw:CARD=S330,DEV=0`

bryanhughes, 2024-04-22:
@dusty-nv- Thanks for all the help. @ms1design - thank you that is good to know. In my case, I am working towards real-time streaming which requires me to use the `pyaudio` python framework.

closed 2025-02-21 by johnnynunez