dusty-nv/jetson-containers #486: cannot build ROS2 docker, Failed building wheel for opencv-contrib-python

samir6577, 2024-04-21:
Hello,

I tried to build the ROS2 docker container ros:foxy-ros-base on my jetson nano after making some changes to the docker file so that I could add some python libraries to my docker container. I have edited the docker file and built the container to add python libraries before using `jetson-containers build ros:foxy-ros-base` a few times, however now when I try to do this I get to 100% and then get the following error message: `  Traceback (most recent call last):
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
      main()
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/usr/local/lib/python3.6/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 262, in build_wheel
      metadata_directory)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 231, in build_wheel
      wheel_directory, config_settings)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
      self.run_setup()
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 268, in run_setup
      self).run_setup(setup_script=setup_script)
    File "/usr/local/lib/python3.6/dist-packages/setuptools/build_meta.py", line 158, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 541, in <module>
      main()
    File "setup.py", line 314, in main
      cmake_source_dir=cmake_source_dir,
    File "/tmp/pip-build-env-4_4wsjoz/overlay/lib/python3.6/site-packages/skbuild/setuptools_wrap.py", line 683, in setup
      cmake_install_dir,
    File "setup.py", line 454, in _classify_installed_files_override
      raise Exception("Not found: '%s'" % relpath_re)
  Exception: Not found: 'python/cv2/mat_wrapper/.*\.py'
  Building wheel for opencv-contrib-python (pyproject.toml): finished with status 'error'
  ERROR: Failed building wheel for opencv-contrib-python
Failed to build opencv-contrib-python
ERROR: Failed to build one or more wheels
The command '/bin/sh -c /tmp/opencv/opencv_install_deps.sh &&     pip3 install --no-cache-dir 	 opencv-contrib-python~=${OPENCV_VERSION}     || /tmp/opencv/build.sh' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/team9jetson/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/team9jetson/jetson-containers/jetson_containers/container.py", line 143, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command 'DOCKER_BUILDKIT=0 docker build --network=host --tag ros:foxy-ros-base-l4t-r32.7.4-opencv --file /home/team9jetson/jetson-containers/packages/opencv/Dockerfile.pip --build-arg BASE_IMAGE=ros:foxy-ros-base-l4t-r32.7.4-numpy --build-arg OPENCV_VERSION="4.5.0" --build-arg OPENCV_PYTHON="4.x" --build-arg CUDA_ARCH_BIN="5.3,6.2,7.2" /home/team9jetson/jetson-containers/packages/opencv 2>&1 | tee /home/team9jetson/jetson-containers/logs/20240421_161329/build/ros_foxy-ros-base-l4t-r32.7.4-opencv.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.`

I tried adding `RUN apt-get update && apt-get install -y \
    libopencv-dev=3.2.0+dfsg-4ubuntu0.1` before ros2_build.sh is run in the dockerfile but this made no difference. Can anyone advise on this please?

dusty-nv, 2024-04-21:
Hi @samir6577 , it has been a while since I built these on JetPack 4, let me try and get back to you.   For now I would recommend just using `dustynv/ros:foxy-ros-base-l4t-r32.7.1` as your base image and installing your desired python packages on top of that, without rebuilding the foxy base container.

samir6577, 2024-04-21:
> Hi @dusty-nv thanks for the response, to build on top of that base image would I run `jetson-containers build dustynv/ros:foxy-ros-base-l4t-r32.7.1` ?

dusty-nv, 2024-04-22:
@samir6577 you would use `FROM dustynv/ros:foxy-ros-base-l4t-r32.7.1` in your own dockerfile and `docker build` it.   I believe I have fixed this in the jetson-containers `dev` branch though with commit [8f7df17](https://github.com/dusty-nv/jetson-containers/commit/8f7df177321596da1f5f5eece68bf08983512412) - so try checking out the dev branch and building `ros:foxy-ros-base` again.

It was failing to compile opencv, even though JetPack 4 could build that version before, so I think some other python packages or toolchains were updated in pip causing it to fail more recently (not a surprise really given how old that version of ubuntu is).  The opencv container now builds again on JetPack 4, and is currently building `ros:foxy-ros-base`

dusty-nv, 2024-04-22:
Ok the `ros:foxy-ros-base` build passed last night on JP6, should be good to go 👍

closed 2025-02-21 by johnnynunez