dusty-nv/jetson-containers #489: Unable to build `diffusion` containers after recent updates to `opencv`

ms1design, 2024-04-25:
Hi @dusty-nv,

Just taking a note that we had a breaking change in `opencv` that negatively impacted the `diffusion` containers breaking the builds:

In `opencv` we have new way of install/building:
https://github.com/dusty-nv/jetson-containers/blob/487dc4ffbd93d59ed7d4450636cabb2fd074c071/packages/opencv/Dockerfile#L25-L29

But this fails hard on `stable-diffusion` container where `invisible-watermark` dependency requires `opencv-python` and cannot find proper distribution.

Same with `stable-diffusion-webui`:
https://github.com/dusty-nv/jetson-containers/blob/487dc4ffbd93d59ed7d4450636cabb2fd074c071/packages/diffusion/stable-diffusion-webui/Dockerfile#L35-L36
```bash
10:44:49 INFO: pip is looking at multiple versions of facexlib to determine which version is compatible with other requirements. This could take a while.
10:44:49 
10:44:49 stderr: ERROR: Ignored the following versions that require a different python version: 0.52.0 Requires-Python >=3.6,<3.9; 0.52.0rc3 Requires-Python >=3.6,<3.9; 0.53.0 Requires-Python >=3.6,<3.10; 0.53.0rc1.post1 Requires-Python >=3.6,<3.10; 0.53.0rc2 Requires-Python >=3.6,<3.10; 0.53.0rc3 Requires-Python >=3.6,<3.10; 0.53.1 Requires-Python >=3.6,<3.10; 0.54.0 Requires-Python >=3.7,<3.10; 0.54.0rc2 Requires-Python >=3.7,<3.10; 0.54.0rc3 Requires-Python >=3.7,<3.10; 0.54.1 Requires-Python >=3.7,<3.10
10:44:49 ERROR: Could not find a version that satisfies the requirement opencv-python (from facexlib) (from versions: none)
10:44:49 ERROR: No matching distribution found for opencv-python
```

Started to fix that in hope it's a quick one – running the `opencv` install/build sh scripts, but I gave up when realised that it starts the `build.sh` from `opencv`. `build.sh` requires the unavailable environment variables (accessible only when building `opencv` cotnainer):
https://github.com/dusty-nv/jetson-containers/blob/487dc4ffbd93d59ed7d4450636cabb2fd074c071/packages/opencv/config.py#L9-L12

ai-and-i, 2024-05-28:
Hi, I have the same issue. @ms1design where you able to solve it or work around it?

ms1design, 2024-05-28:
Unfortunately, no, but I haven't tried yet tbh. I have this on my to-do list for when I have more time available.

ai-and-i, 2024-05-28:
I think I fixed it in #541

dusty-nv, 2024-05-29:
Thanks @ai-and-i , took a deeper look and fixed it another way in https://github.com/dusty-nv/jetson-containers/commit/e0368bf4d0ce319171596600f0bdae3b4a7be045

It was related to the CUDA-enabled opencv wheels being called `opencv-contrib-python` not `opencv-python`, but yet I was blacklisting `opencv-python` on the pip server from mirroring PyPi (so it wouldn't install the non-CUDA versions).  So I disabled that one since it was creating errors during pip installs, and instead reinstalled the OpenCV-CUDA binaries near the end of the build.

ai-and-i, 2024-05-29:
Perfect, thanks for a quick fix!

ms1design, 2024-06-19:
Fixed in e0368bf4d0ce319171596600f0bdae3b4a7be045, closing.

closed 2024-06-19 by ms1design