dusty-nv/jetson-containers #508: Workflow for migrating from Jetpack 6 DP to Jetpack 6 GA ?

telemetrieTP23, 2024-05-09:
After migrating from DP to GA with this steps: 
https://docs.nvidia.com/jetson/archives/jetpack-archived/jetpack-60/install-setup/index.html#upgrade-jetpack

The docker commands are not working anymore:

 jetson-containers run --name ollama $(autotag ollama)

docker run -it --rm --network=host --add-host=host.docker.internal:host-gateway -v open-webui:/app/backend/data ghcr.io/open-webui/open-webui:main

-> 
docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.

Does the update broke docker ?

If there is an easy way to fix this, please help. greetings

TadayukiOkada, 2024-05-09:
What does this show?:
`sudo systemctl status docker`

martincerven, 2024-05-09:
Does this [iptables fix ](https://forums.developer.nvidia.com/t/any-software-update-to-jetpack-6-0-causes-docker-to-fail/283910/9)work? I also had to do that first time I ran containers on Jetsons ( JP 6.x )

telemetrieTP23, 2024-05-09:
You are right my posting was easy to misinterpret, i used 2 terminals, each of them for one command.

`sudo systemctl status docker says:

```
× docker.service - Docker Application Container Engine
     Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-05-09 12:32:03 CEST; 5min ago
TriggeredBy: × docker.socket
       Docs: https://docs.docker.com
    Process: 2288 ExecStart=/usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.sock (code=exited, status=1/FAILURE)
   Main PID: 2288 (code=exited, status=1/FAILURE)
        CPU: 220ms

Mai 09 12:32:03 orin-desktop systemd[1]: docker.service: Scheduled restart job, restart counter is at 3.
Mai 09 12:32:03 orin-desktop systemd[1]: Stopped Docker Application Container Engine.
Mai 09 12:32:03 orin-desktop systemd[1]: docker.service: Start request repeated too quickly.
Mai 09 12:32:03 orin-desktop systemd[1]: docker.service: Failed with result 'exit-code'.
Mai 09 12:32:03 orin-desktop systemd[1]: Failed to start Docker Application Container Engine.
```
I can search the docker issue for my one, but asked here if dusty or some other nvidia member has tips for upgrading this way ( https://docs.nvidia.com/jetson/archives/jetpack-archived/jetpack-60/install-setup/index.html#upgrade-jetpack ) , because it was recommended by nvidia 
`

telemetrieTP23, 2024-05-09:
> Does this [iptables fix ](https://forums.developer.nvidia.com/t/any-software-update-to-jetpack-6-0-causes-docker-to-fail/283910/9)work? I also had to do that first time I ran containers on Jetsons ( JP 6.x )

YESS. It worked, the link to your solution should be posted in the official nvidia update guide.

martincerven, 2024-05-13:
Thanks, if that solved your problem, mark the issue as closed 😁
> YESS. It worked, the link to your solution should be posted in the official nvidia update guide.

closed 2024-05-13 by telemetrieTP23