dusty-nv/jetson-containers #509: Building llama_cpp container does not produce the compiled C++ version of llama.cpp executables

amasciotta, 2024-05-09:
Hello! I am using the prebuilt container [dustynv/llama_cpp](https://hub.docker.com/r/dustynv/llama_cpp/tags), which contains the built C++ executables inside `/opt/llama.cpp`.

However, trying to rebuild it from scratch to get some fixes from the main repo, I obtain a container that has no `/opt/llama.cpp` folder.

I suspect that #422 broke something...

dusty-nv, 2024-05-09:
Hi @amasciotta, yes it builds llama_cpp_python from https://github.com/abetlen/llama-cpp-python which has llama.cpp as submodule.  For now if you need to retain the C++ stuff, you just may want to build it, it is straightforward to compile these days with `-DLLAMA_CUBLAS=on -DLLAMA_CUDA_F16=1`

amasciotta, 2024-05-10:
I will definitely try to build it myself. Any other suggestion on how to set compile options to work best on Jetson Xavier and Orin?

Finally, a note: `LLAMA_CUBLAS` is [deprecated](https://github.com/ggerganov/llama.cpp/blob/25c6e82e7a1ad25a42b0894e87d9b5c557409516/Makefile#L426), in favour of `LLAMA_CUDA`.

hillct, 2024-07-16:
I'm not entirely convinced my local build issue on an AGX Orin 32GB running Jetpack 6, with jetson-containers pulled today #commit:e77e05ebcbbb15e315bbf8b5bcd160946995a12c  is the same as described in this ticket, but figured I'd supply my buildlog here nonetheless rather than open what is likely to b a duplicate ticket.
[agx_orin_llama_cpp_buildlog.txt](https://github.com/user-attachments/files/16255100/agx_orin_llama_cpp_buildlog.txt)

dusty-nv, 2024-07-16:
> `OSError: libnvrm_gpu.so: cannot open shared object file: No such file or directory`

@hillct that looks like a different issue, like your docker engine or nvidia runtime is misconfigured.  It seems unable to access GPU drivers during the build - do you have your default docker runtime set to nvidia, like shown here?

https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

Alternatively, are you able to do things like `import tensorrt` inside l4t-jetpack container?

hillct, 2024-07-17:
> @hillct that looks like a different issue, like your docker engine or nvidia runtime is misconfigured. It seems unable to access GPU drivers during the build - do you have your default docker [...]

The nvidia runtime was defined in that I was always able to run CUDA-enabled containers but it had always been my habit to specify the nvidia runtime on the docker command line ie: `--runtime nvidia`. The missing property was "default-runtime" which would have removed the command line arg requirement. Presumably on of the scripted build steps omitted the command line arg. It's worth noting that other containers locally built using your jetson-containers build scripts seem to build successfully, notably here, for example ollama, which upon review does explicitly specify the nvidia runtime on the docker command line, so the absence of the explicit specification is llama_cpp specific not at the jetson-conainers script level. Whether or not i should be 'fixed' for this build script, is debatable, but there's a good argument for consistency across  all jetson-containers build scripts.

As I write this, a new llama_cpp build has completed successfully, so the specification of default-runtime was indeed the issue. Thanks!

PS: Looking now at the final docker run command, the `--runtime nvidia` arg is specified so it must have related only to the intermediate build step.

> Alternatively, are you able to do things like `import tensorrt` inside l4t-jetpack container?

dusty-nv, 2024-07-17:
`docker build` does not accept the `--runtime` flag, but does honor the default-runtime being set in `/etc/docker/daemon.json` (unless you are using buildkit, which is why I disable that).  So for containers that use CUDA during the build, those need it.  The tests are run with `docker run --runtime nvidia` at the conclusion of each dockerfile being built.