dusty-nv/jetson-containers #511: XTTS build

budzianowski, 2024-05-09:
Thanks for the great work! Three questions:
1. For the version build for 35.2.1 do I have to rebuild python3.9 (this is coqui requirement)? For JP5 what's the best way to go about it?
2. https://github.com/dusty-nv/jetson-containers/blob/aef3d09267c0b8564c6f3c18dc59873ce0a179d4/packages/audio/xtts/test_voices.py#L12C1-L12C57 - the path to the models have been changed (https://huggingface.co/coqui/XTTS-v2/tree/main)
3. If I understand the quantization happens during the setup?

dusty-nv, 2024-05-09:
@budzianowski I'm not sure that I ever got it working on JetPack 5 and only enable it in my stuff on JetPack 6, and even so found XTTS to be too slow in practice (even on AGX), so I changed to using Piper TTS.

budzianowski, 2024-05-09:
Thanks for the heads up - I guess the <200ms latency claims were exaggerated. I will have a look at Piper.

closed 2024-05-09 by budzianowski

Tetsuo-tek, 2024-05-18:
Thanks for jetson-containers, it is very useful!

I would-like to run also xtts container  on my jetson-orin-AGX. 

$ jetson-containers run $(autotag xtts)
Namespace(packages=['xtts'], prefer=['local', 'registry', 'build'], disable=[''], user='dustynv', output='/tmp/autotag', quiet=False, verbose=False)
-- L4T_VERSION=36.2.0  JETPACK_VERSION=6.0  CUDA_VERSION=122.0
-- Finding compatible container image for ['xtts']

Couldn't find a compatible container for xtts, would you like to build it? [y/N] y
-- Package onnxruntime:1.17 has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package onnxruntime:1.17-builder has missing dependencies, disabling...  ("couldn't find package:  cuda")
-- Package tensorflow2 has missing dependencies, disabling...  ("couldn't find package:  cuda")

[..]

and more and more dependencies not found.

I need to use xtts_v2 from Hugging-Face, and I cannot substitute it with piper.
I'm new to jetson and docker. Can you help me to build this container or how to go on the right way?

Horschig, 2025-02-23:
@Tetsuo-tek did you get it to work in the end? I am considering getting a nano super and am interested to try out cocqui xtts v2.