dusty-nv/jetson-containers #517: Regarding the issue of mutual interference between the speaker and the microphone in LlamaSpeak

lq0104, 2024-05-11:
Hello, I would like to ask a question, although it may not be within the scope of this repository. I wanted to understand how to deal with a problem when using LlamaSpeak. The sound emitted by the speaker is being picked up by the microphone, resulting in a lot of interference during the conversation. Even when I am not speaking, the chat robot keeps talking. I wanted to know how you handle this issue. Do you move the microphone and speaker further apart? However, it seems that sometimes this problem still occurs.

dusty-nv, 2024-05-12:
Hi @lq0104, I use a microphone/speaker combo with integrated noise cancellation (like the [Anker PowerConf S3](https://us.ankerwork.com/products/a3302?_pos=5&_sid=a228ea1b1&_ss=r), or headsets I never have a problem with either)

lq0104, 2024-05-13:
Thank you very much for your kind response, I will also give it a try.

closed 2024-05-13 by lq0104