dusty-nv/jetson-containers #529: Guide for MLC is outdated due to MLC updates

ramyadhadidi, 2024-05-15:
Hello, seems like the guide found here https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/mlc/README.md is outdated.  Specifically, `mlc_llm.build` cannot be found.

```
./run.sh $(./autotag mlc) \
  python3 -m mlc_llm.build \
    --model Llama-2-7b-chat-hf \
    --quantization q4f16_ft \
    --artifact-path /data/models/mlc/dist \
    --max-seq-len 4096 \
    --target cuda \
    --use-cuda-graph \
    --use-flash-attn-mqa
```

I did some workaround but not sure if that is what is intended. I think it should be replaced by that has a custom SNAPSHOT_ID:
```
 python3 -m mlc_llm convert_weight /data/models/huggingface/models--meta-llama--Llama-2-7b-chat-hf/snapshots/SNAPSHOT_ID --model-type llama --quantization q4f16_ft --output /data/models/mlc/dist --device cuda
```

dusty-nv, 2024-05-16:
Ah thanks @ramyadhadidi , yes I have various versions of MLC floating around, and the latest was after their transition from `mlc_llm.build` to `mlc_llm convert_weight` way.  It seems like I pushed the builder but not the deployment container - which version of JetPack-L4T are you on?

ramyadhadidi, 2024-05-16:
I'm on the latest one
L4T_VERSION=36.3.0  JETPACK_VERSION=6.0  CUDA_VERSION=12.2

closed 2025-02-21 by johnnynunez