dusty-nv/jetson-containers #530: Can't build CUDA=11.8 ROS2 Humble Image

jj-quantumdrones, 2024-05-17:
I have an NVIDIA Jetson Orin NX.

I am trying to build an image with the following configuration:
- CUDA=11.8
- ROS2 Humble

I am going to be working in python and cpp.

I run the following command: 

```bash
$ CUDA_VERSION=11.8 jetson-containers build --name=ai-dev ros:humble-desktop
```

I get the following output:

<details>
  <summary>Click to expand output</summary>

```txt
DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            BuildKit is currently disabled; enable it by removing the DOCKER_BUILDKIT=0
            environment-variable.

Sending build context to Docker daemon  19.97kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
r35.4.1: Pulling from nvidia/l4t-jetpack
2378679266ac: Pulling fs layer
fbe64e031ca2: Pulling fs layer
bc84636d073b: Pulling fs layer
4c8a78053e02: Pulling fs layer
c98a245e07e2: Pulling fs layer
54e58b7fa9fd: Pulling fs layer
bd70e595c704: Pulling fs layer
70a7c3288501: Pulling fs layer
0cdb0dfb035f: Pulling fs layer
a4c3570418d1: Pulling fs layer
019c19b28a48: Pulling fs layer
ea8790418f30: Pulling fs layer
a9dca93572f1: Pulling fs layer
1a4f4ffe916b: Pulling fs layer
b1285e2f3ec3: Pulling fs layer
2bd4b945552e: Pulling fs layer
a5c388f5daf7: Pulling fs layer
d9fa505958b2: Pulling fs layer
5f5b8fb8b6a8: Pulling fs layer
1b798dc9c4ba: Pulling fs layer
b48c074786ef: Pulling fs layer
30eefdd3bb15: Pulling fs layer
c98a245e07e2: Waiting
54e58b7fa9fd: Waiting
bd70e595c704: Waiting
70a7c3288501: Waiting
0cdb0dfb035f: Waiting
a4c3570418d1: Waiting
019c19b28a48: Waiting
ea8790418f30: Waiting
a9dca93572f1: Waiting
1a4f4ffe916b: Waiting
b1285e2f3ec3: Waiting
2bd4b945552e: Waiting
4c8a78053e02: Waiting
a5c388f5daf7: Waiting
d9fa505958b2: Waiting
5f5b8fb8b6a8: Waiting
1b798dc9c4ba: Waiting
b48c074786ef: Waiting
30eefdd3bb15: Waiting
bc84636d073b: Verifying Checksum
[... 861 lines cut ...]
curl: (23) Failed writing body (0 != 16384)
The command '/bin/sh -c /tmp/install_python.sh' returned a non-zero code: 127
```

It looks like python is failing to install correctly?

I'm not sure what the state of my image and containers are:

```bash
qsai@ai-board:~/data/jacques/code$ sudo docker image ls
REPOSITORY                   TAG                           IMAGE ID       CREATED         SIZE
<none>                       <none>                        cba51d6a5507   3 minutes ago   10GB
ai-dev                       l4t-r35.4.1-cuda              41e3fc7bbf4e   3 minutes ago   10GB
ai-dev                       l4t-r35.4.1-cudnn             41e3fc7bbf4e   3 minutes ago   10GB
ai-dev                       l4t-r35.4.1-build-essential   f58f86bf560e   3 minutes ago   10GB
nvcr.io/nvidia/l4t-jetpack   r35.4.1
```

```bash
qsai@ai-board:~/data/jacques/code$ sudo docker ps -a
CONTAINER ID   IMAGE          COMMAND                  CREATED         STATUS                       PORTS     NAMES
3f7fa8760993   cba51d6a5507   "/bin/sh -c /tmp/ins…"   4 minutes ago   Exited (127) 3 minutes ago             eager_villani
```

I would greatly appreciate some advice on what to do next to get this set up. Thank you.

jj-quantumdrones, 2024-05-17:
PS

Once the image is built on my Orin I'm trying to run my dev container with the following setup:

```
.devcontainer/
    devcontainer.json
    Dockerfile
```

devcontainer.json
```json
{
    "name": "AI Development Container",
    // "image": "ai-dev:l4t-r35.4.1-cuda",
    "image": "cba51d6a5507",
    "runArgs": [
        "--gpus",
        "all"
    ],
    "mounts": [
        "source=${localWorkspaceFolder},target=/workspace,type=bind,consistency=cached"
    ],
    "workspaceFolder": "/workspace",
    "customizations": {
        "vscode": {
            "extensions": [
                "ms-python.python"
            ]
        }
    }
}
```

Dockerfile:
```Dockerfile
FROM ai-dev

# Install any additional dependencies or perform further setup if needed
```

dusty-nv, 2024-05-17:
Hi @jj-quantumdrones, on JetPack 5 I still use l4t-jetpack as the base container (i.e. it already comes with the default version of CUDA 11.4, cuDNN, TensorRT, ect) and have not done custom CUDA versions for JetPack 5.  Sorry that I only have time to support that for JetPack 6.  Since you have Orin, could you run JetPack 6 on that?

jj-quantumdrones, 2024-05-17:
Hi @dusty-nv thank you for your time to answer my post.

Unfortunately I cannot run JetPack 6 we have to use JetPack 5 for our project. But from some testing today without docker trying to run our app straight on the Orin board it looks like I might be fine to run CUDA 11.4 and I may not need CUDA 11.8.

I tried running `$ jetson-containers build --name=ai-dev ros:humble-desktop` without the CUDA 11.8 flag and still ran into the same python installation error though I'm not sure how to get past...

dusty-nv, 2024-05-17:
OK thanks for letting me know @jj-quantumdrones , the past couple days I have been getting nano_llm to build on top of ros:humble container (see [forum thread](https://forums.developer.nvidia.com/t/combining-jetson-containers/292883) about this), and will also go back and check it on JP5 too 👍

jj-quantumdrones, 2024-05-21:
Thank you so much! I will try to go through the setup today again with my Orin NX.

jj-quantumdrones, 2024-05-24:
@dusty-nv I am now running `jetson-containers build --name=ai-dev ros:humble-desktop` and I get the following output logs, they are very long so I am attaching here as `.txt` files: 

- [output.txt](https://github.com/dusty-nv/jetson-containers/files/15440240/output.txt)
- [output2.txt](https://github.com/dusty-nv/jetson-containers/files/15440241/output2.txt)
- [output3.txt](https://github.com/dusty-nv/jetson-containers/files/15440242/output3.txt)
- [output4.txt](https://github.com/dusty-nv/jetson-containers/files/15440243/output4.txt)

They all end the same as follows, exit with status 0 which appears positive.

```bash
-- Building container ai-dev:l4t-r35.4.1-ros_humble-desktop

DOCKER_BUILDKIT=0 docker build --network=host --tag ai-dev:l4t-r35.4.1-ros_humble-desktop \
--file /home/picard/Code/Containers/jetson-containers/packages/ros/Dockerfile.ros2 \
--build-arg BASE_IMAGE=ai-dev:l4t-r35.4.1-cmake \
--build-arg ROS_VERSION="humble" \
--build-arg ROS_PACKAGE="desktop" \
/home/picard/Code/Containers/jetson-containers/packages/ros \
2>&1 | tee /home/picard/Code/Containers/jetson-containers/logs/20240524_131237/build/ai-dev_l4t-r35.4.1-ros_humble-desktop.txt; exit ${PIPESTATUS[0]}
```

But from the same `output.txt` file this part looks suspicious:

```bash
[91m/usr/bin/vcs:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
[0m.E...................................................................................................................................................................................................................................................................................................................................
```

Moving on to try to inspect the status of my docker images:

When I run `docker images` to see the status of my images, the final image `ai-dev_l4t-r35.4.1-ros_humble-desktop` is missing.

```bash
picard@picard-desktop:~/Code/Containers$ docker images
REPOSITORY                   TAG                           IMAGE ID       CREATED          SIZE
<none>                       <none>                        9499b28c2b2e   39 minutes ago   11.1GB
<none>                       <none>                        5138c9a57ae4   3 hours ago      11.1GB
ai-dev                       l4t-r35.4.1-cmake             9aca54a6fd47   3 hours ago      11.1GB
ai-dev                       l4t-r35.4.1-opencv_deb        59d492886a25   3 hours ago      11.1GB
ai-dev                       l4t-r35.4.1-numpy             3f887eef4629   3 hours ago      10.2GB
ai-dev                       l4t-r35.4.1-python            a037d2d19cad   3 hours ago      10.1GB
ai-dev                       l4t-r35.4.1-tensorrt          a037d2d19cad   3 hours ago      10.1GB
ai-dev                       l4t-r35.4.1-build-essential   829f0d2664c0   3 hours ago      10GB
ai-dev                       l4t-r35.4.1-cuda              dd261a37448c   3 hours ago      10GB
ai-dev                       l4t-r35.4.1-cudnn             dd261a37448c   3 hours ago      10GB
nvcr.io/nvidia/l4t-jetpack   r35.4.1                       5c923ac521a3   10 months ago    9.71GB
```

Notice there are two unnamed untagged images...

So then I look at the logs here: `~/Code/Containers/jetson-containers/logs/20240524_141042/build/ai-dev_l4t-r35.4.1-ros_humble-desktop.txt` for building this last container called `ai-dev_l4t-r35.4.1-ros_humble-desktop`

See: [ai-dev_l4t-r35.4.1-ros_humble-desktop.txt](https://github.com/dusty-nv/jetson-containers/files/15440327/ai-dev_l4t-r35.4.1-ros_humble-desktop.txt)

In here I see:

```bash
[91m/usr/bin/vcs:6: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import load_entry_point
[0m.................................E.E...................................................................................E..............................................................................................................................E..............................................................................
```

I assume those `E`s to be errors in some sort of progress bar?

I also get this at the end:

```bash
HEAD is now at 8364932 Trimmed the branch to only the vision_opencv sub directory
=== src/yaml_cpp_vendor (git) ===
Cloning into '.'...
Note: switching to 'release/humble/yaml_cpp_vendor/8.0.2-1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at fbd4968 Rebase from 'upstream'
The command '/bin/bash -c ./ros2_build.sh' returned a non-zero code: 1
```

So it looks like `ros2_build.sh` is failing...

aapatni, 2024-09-26:
Hi @jj-quantumdrones were you able to root cause this error?

closed 2025-02-21 by johnnynunez