dusty-nv/jetson-containers #531: How to use 'nsys' in mlc container

Louym, 2024-05-17:
I'm using a pre-built MLC container on my ARM64 (aarch64) platform with an Orin GPU. I've successfully benchmarked the speed of LLM as per the provided documentation. However, I'm facing an issue where I can only trace CPU activities when attempting to analyze with nsys. Since GPU information is crucial for our analysis, I need to resolve this. How can I configure nsys within the container to include GPU tracing?
[image]

dusty-nv, 2024-05-17:
Hi @Louym, firstly I'm glad that you were able to get MLC running and benchmarked.  I haven't used nsys inside container before and am not sure if there is special setup or tools needed installed inside the container.  You might just want to start by profiling a simple CUDA app in a simple CUDA container if you haven't already.  Otherwise, you can find the wheels from here (which you can install outside container):  http://jetson.webredirect.org/jp5/cu114

Louym, 2024-05-17:
> Hi @Louym, firstly I'm glad that you were able to get MLC running and benchmarked. I haven't used nsys inside container before and am not sure if there is special setup or tools needed installed [...]

Thank you very much for your prompt response. I will try this website later, and I will get back to you if I have any issues.

dusty-nv, 2024-05-21:
Hmm I haven't done this outside of container, but it would seem that the MLC version of the wheel does not correspond to the TVM version of the wheel.  In the containers I lock the package versions to make sure the matching ones get installed, which you can see those versions here in its config.py:

https://github.com/dusty-nv/jetson-containers/blob/5ba535905ff275e02ff779112d351a754aece94f/packages/llm/mlc/config.py#L31

In general, I would try and replicate the install just how I have done it in the container.  It looks like you are using conda and I don't think that should impact it but not sure?

Louym, 2024-05-24:
I tried again and find that I can use 'nsys profile -stats true' to see the summary of GPU kernels. But I also meet some issues when building tvm-unity(or relax) from source, although I have: 
set(USE_CUDA  ON)
set(USE_FLASHINFER ON)
set(FLASHINFER_CUDA_ARCHITECTURES 87)
set(CMAKE_CUDA_ARCHITECTURES 87)
[image]
It would be great if you could help me solve this problem.

Louym, 2024-05-30:
I manage to see the nsys timeline results by compiling models in containers and then benchmarking it in the environment built with the website http://jetson.webredirect.org/jp6/cu122 using jp 6 and cuda 12.2 out of containers.

closed 2024-05-30 by Louym