dusty-nv/jetson-containers #532: Cannot Reproduced Published Benchmarking Result on Jetson AI Lab Webpage

ramyadhadidi, 2024-05-20:
Hello,

I've been trying to reproduce the benchmarking results published on the Jetson AI Lab webpage ([here](https://www.jetson-ai-lab.com/benchmarks.html)).

First of all, it is unclear which Jetson Orin models are utilized, i.e., AGX: [32GB, 64GB, or Industrial] / Nano: [4GB or 8GB].

Second, I'm having a hard time matching the reported numbers. I get 3-4x lower performance. I followed the `MLC` guide [here](https://github.com/dusty-nv/jetson-containers/tree/master/packages/llm/mlc) with some fixes (see https://github.com/dusty-nv/jetson-containers/issues/529). For instance: 
- For Llama2-7b, I get ~19 tokens/sec (versus 47).
- For Gemma, I get ~23 tokens/sec (versus 75).
- For Phi2, MLC just says it does not support Phi2.

I have an AGX Orin 32GB, so partially these lower numbers might be related to the lower memory and TOPS difference of the two AGX variants (~1.4x). However, I don't think this is the main reason. None of the above models are bottlenecked by the smaller memory size of the AGX 32GB, and the TOPS difference is not significant considering that Attention is a memory-bound operation (memory bandwidth of both models are same).

dusty-nv, 2024-05-20:
Hi Ramyad, this was with AGX Orin 64GB in MAX-N power mode.  Other folks typically have to change to this power mode (using nvpmodel tool) then they get the same 47 tokens/sec with Llama-2-7B for example.  IIRC the AGX 32GB has fewer cores not just memory capacity.

ramyadhadidi, 2024-05-20:
Hi Dustin, thanks. You are right, I am on 30W mode. I will close this issue and reopen it if I cannot produce the result that is almost similar to AGX 64GB.

closed 2024-05-20 by ramyadhadidi

luhuaei, 2024-09-23:
Hi, Dustin @dusty-nv 

I encountered the same problem as mentioned above. I'm using a Jetson AGX Orin 64 GB device, running `Llama-2-13b-chat-hf-q4f16_1-MLC` [huggingface link](https://huggingface.co/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC) in the mlc `dustynv/mlc:0.1.1-r36.3.0` container. However, the test results show only `20.6 tokens/s`, which is different from the speed given in the official [benchmark](https://www.jetson-ai-lab.com/benchmarks.html). Could you please tell me if there's something wrong with my configuration? Thank you for your guidance.

some output
```
root@nvidia-desktop:/workspace/packages/llm/mlc# python3 benchmark.py --model /workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC/ --max-new-tokens 128 --max-num-prompts 4 --prompt /data/prompts/completion_16.json --save /workspace/myresults/mlc.csv
failed to import ChatModule from mlc_chat (No module named 'mlc_chat')
trying to import ChatModule from mlc_llm instead...
Namespace(model='/workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC/', model_lib_path=None, prompt=['/data/prompts/completion_16.json'], chat=False, streaming=False, max_new_tokens=128, max_num_prompts=4, save='/workspace/myresults/mlc.csv')
-- loading /workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC/
[2024-09-23 08:55:01] INFO auto_device.py:76: Found device: cuda:0
[2024-09-23 08:55:04] INFO auto_device.py:85: Not found device: rocm:0
[2024-09-23 08:55:06] INFO auto_device.py:85: Not found device: metal:0
[2024-09-23 08:55:08] INFO auto_device.py:85: Not found device: vulkan:0
[2024-09-23 08:55:11] INFO auto_device.py:85: Not found device: opencl:0
[2024-09-23 08:55:11] INFO auto_device.py:33: Using device: cuda:0
[2024-09-23 08:55:11] INFO chat_module.py:379: Using model folder: /workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC
[2024-09-23 08:55:11] INFO chat_module.py:380: Using mlc chat config: /workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC/mlc-chat-config.json
[2024-09-23 08:55:11] INFO chat_module.py:781: Model lib not found. Now compiling model lib on device...
[2024-09-23 08:55:11] INFO jit.py:35: MLC_JIT_POLICY = ON. Can be one of: ON, OFF, REDO, READONLY
[2024-09-23 08:55:11] INFO jit.py:117: Using cached model lib: /root/.cache/mlc_llm/model_lib/f59ba09a21ac1031e887a08ef6fc1641.so
[2024-09-23 08:55:13] INFO model_metadata.py:96: Total memory usage: 7602.82 MB (Parameters: 6982.82 MB. KVCache: 0.00 MB. Temporary buffer: 620.00 MB)
[2024-09-23 08:55:13] INFO model_metadata.py:105: To reduce memory usage, tweak `prefill_chunk_size`, `context_window_size` and `sliding_window_size`
[08:55:13] /opt/mlc-llm/cpp/llm_chat.cc:568: Warning: Legacy conversation template detected. It will be deprecated in the future. Please regenerate mlc-chat-config.json with the latest version

PROMPT:  Once upon a time, there was a little girl who loved to read.

She spent hours every day curled up with a good book, exploring magical worlds and meeting amazing characters. As she grew older, she began to write her own stories, pouring her imagination and creativity onto the page. She wrote about magic and adventure, about love and loss, about the human condition and the power of the human spirit. And as she wrote, she discovered that the world of words was just as magical as the world of dreams.

This is the story of my life. I have always been a lover of words, a seeker of magic, and a belie

/workspace/mymodels/mlc-ai/Llama-2-13b-chat-hf-q4f16_1-MLC/:  input=16 output=128 prefill_time 0.217 sec, prefill_rate 73.9 tokens/sec, decode_time 6.214 sec, decode_rate 20.6 tokens/sec
```

jetson device release
```
$ jetson_release 
Software part of jetson-stats 4.2.10 - (c) 2024, Raffaello Bonghi
Model: Jetson AGX Orin - Jetpack 5.1.2 [L4T 35.4.1]
NV Power Mode[0]: MAXN
Serial Number: [XXX Show with: jetson_release -s XXX]
Hardware:
 - P-Number: xxxxxxx
 - Module: NVIDIA Jetson AGX Orin (64GB ram)
Platform:
 - Distribution: Ubuntu 20.04 focal
 - Release: 5.10.120-tegra
jtop:
 - Version: 4.2.10
 - Service: Active
Libraries:
 - CUDA: 11.4.315
 - cuDNN: 8.6.0.166
 - TensorRT: 8.5.2.2
 - VPI: 2.3.9
 - Vulkan: 1.3.204
 - OpenCV: 4.5.4 - with CUDA: NO
```

jtop screeshot on running
[image]

dusty-nv, 2024-09-23:
@luhuaei it looks like you are on JetPack 5 whereas those benchmarks are with JetPack 6 and CUDA 12.2, and for llama using `dustynv/mlc:0.1.0-r36.2.0`

luhuaei, 2024-09-24:
> @luhuaei it looks like you are on JetPack 5 whereas those benchmarks are with JetPack 6 and CUDA 12.2, and for llama using `dustynv/mlc:0.1.0-r36.2.0`

Thanks, I try to latest version.