dusty-nv/jetson-containers #535: local_llm: no way to change or disable ICE server for webrtc

TadayukiOkada, 2024-05-22:
I'd like to run live llava completely locally on Jetson including a web browser.
However, if I turn off wifi before starting live llava, the video won't play on the browser.
If I turn off wifi after starting it, the video continues to play.
I'm guessing this is because the default ICE server is set to google's STUN server.
I tried to find a way to change or disable it but it seems there's no way to change it using option flags right now.
I'm willing to create a PR for this, but not sure what's the cleanest way to set it.

dusty-nv, 2024-05-22:
@TadayukiOkada it looks like these can be disabled like this:  https://stackoverflow.com/questions/30742431/webrtc-on-isolated-lan-without-ice-stun-turn-server

That is on the javascript/client side.  However on the server side, I'm not sure if gstreamer webrtcbin will work without STUN server.  Here are the references to it in jetson_utils:   https://github.com/search?q=repo%3Adusty-nv%2Fjetson-utils%20stun&type=code

I would get jetson_utils working first without STUN, and then the javascript in NanoLLM can be updated to reflect it.  It may be tricky however to support both ways.

manhntm3, 2024-05-24:
Hi @dusty-nv, 

I've tried to disable STUN server on both javascript and gstreamer by commenting out the STUN part and it works when the server(gstreamer) side is not connected to Internet. 

However, it still fails to connect when viewed from a PC (same LAN) that doesn't connect to the Internet.

POLROTY, 2024-06-05:
@manhntm3 I am having the same problem, I can't view the video from an other device on the lan if not connected to internet. My jetson also crashes because the queue frames fills up all the RAM. Have you found a fix?

Rz-Rz, 2024-06-10:
@manhntm3 I am also having the same problem. This is very problematic.

dusty-nv, 2024-06-14:
Can you try this attempted fix from jetson-utils?   https://github.com/dusty-nv/jetson-utils/commit/72c4e095f0575091992a5803d79af08e138e3369

```
video-viewer /dev/video0 webrtc://@:8554/output --stun-server disabled
```

Then navigate your browser to `http://JETSON_IP:8554`

When it is set to disabled, I set the ice-servers in the built-in HTML/Javascript to empty array as suggested here:

https://groups.google.com/g/discuss-webrtc/c/s7WsDPjHIj4/m/v8a6jPWtF8QJ

It worked for me both with it disabled and the default server (`stun.l.google.com:19302`)

The web-side changes are only for the jetson-utils built-in debug viewer on port 8554 so far.

manhntm3, 2024-06-24:
For me, it only works when the computer that views Jetson's webpage is connected to the Internet. 
If we disable the Internet from the laptop that runs the browser and see Jetson's webpage through port 8554, it still doesn't work.
[image]

dusty-nv, 2024-06-24:
Can you confirm in the clientside HTML/Javascript that the ICE stun servers are in fact being set to empty?

Screenshot.2024-06-24.at.13.15.52.png (view on web)<https://github.com/dusty-nv/jetson-containers/assets/53960609/a21f44f0-ea48-4201-9883-3037224456f7>

POLROTY, 2024-06-25:
It seems to be working for me, I made a new connection mechanism client side directly with a websocket and with a local adapter-latest.js, as I think this import doesn't work without access to the internet : line 44 of WebrtcServer.cpp <script type='text/javascript' src='https://webrtc.github.io/adapter/adapter-latest.js'></script>

manhntm3, 2024-06-26:
> Can you confirm in the clientside HTML/Javascript that the ICE stun servers are in fact being set to empty?
> […](#)

Yes, I can confirm the clientside HTML/Javascript is being set to empty. 
[image]

What I did to stop Internet from my laptop and keep the LAN connection is change to static IP and empty the router box. It might affect only on my side. On your side, does the WebRTC video work when you disable Internet on the client machine or both Jetson and client?

POLROTY, 2024-07-04:
Update, I tried again by connecting the jetson and and other computer with an ethernet cable and without an internet connection on either side and it doesn't seem to work offline in this configuration...
I have this in the logs: 
root@ubuntu:/jetson-utils/build/aarch64/bin# video-viewer /dev/video0 webrtc://@:8554/output --stun-server disabled
[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: USB3.0 Capture: USB3.0 Capture
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"USB3.0\ Capture:\ USB3.0\ Capture", v4l2.device.bus_info=(string)usb-3610000.xhci-3.4, v4l2.device.version=(uint)330432, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 22 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 10/1 };
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)960, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)1360, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [7] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [8] video/x-raw, format=(string)YUY2, width=(int)720, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [9] video/x-raw, format=(string)YUY2, width=(int)720, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [10] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [11] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 20/1, 10/1 };
[gstreamer] [12] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [13] image/jpeg, width=(int)1280, height=(int)1024, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [14] image/jpeg, width=(int)1280, height=(int)960, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [15] image/jpeg, width=(int)1360, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [16] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [17] image/jpeg, width=(int)1024, height=(int)768, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [18] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [19] image/jpeg, width=(int)720, height=(int)576, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [20] image/jpeg, width=(int)720, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] [21] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 60/1, 50/1, 30/1, 20/1, 10/1 };
[gstreamer] gstCamera -- selected device profile:  codec=raw format=yuyv width=1280 height=720 framerate=30
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, framerate=30/1 ! appsink name=mysink sync=false
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- codecType:  cpu
  -- width:      1280
  -- height:     720
  -- frameRate:  30
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- stunServer  disabled
------------------------------------------------
[... cut ...]
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> srtpenc1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> dtlssrtpenc1
video-viewer:  captured 100 frames (1280x720)
video-viewer:  captured 125 frames (1280x720)
video-viewer:  captured 150 frames (1280x720)
video-viewer:  captured 175 frames (1280x720)
video-viewer:  captured 200 frames (1280x720)
video-viewer:  captured 225 frames (1280x720)
video-viewer:  captured 250 frames (1280x720)
video-viewer:  captured 275 frames (1280x720)

stephemer, 2025-02-19:
@POLROTY Did you ever find a solution for this?

Rz-Rz, 2025-02-21:
Don't think this issue should be closed

T-omar-For-NOV, 2025-03-15:
any updates about the issue please ?