dusty-nv/jetson-containers #537: Can't build ROS2 Humble Image

mswanson24, 2024-05-23:
I have an NVIDIA Jetson Orin NX.

I am trying to build an image with the following configuration:

- ROS2 Humble

I am going to be working in python and cpp.

FYI this is a continuation of [Can't build CUDA=11.8 ROS2 Humble Image #530](https://github.com/dusty-nv/jetson-containers/issues/530#issue-2302275837)

We are using Jetpack 5.1 and CUDA 11.4

I run the following command:
```shell
jetson-containers build --name=ai-dev ros:humble-desktop
```
I get the following Error:
```
-- Testing container ai-dev:l4t-r35.3.1 (opencv:4.5.0-deb/test.py)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/orin_nx/Desktop/software/jetson-containers/packages/opencv:/test \
--volume /home/orin_nx/Desktop/software/jetson-containers/data:/data \
--workdir /test \
ai-dev:l4t-r35.3.1 \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/orin_nx/Desktop/software/jetson-containers/logs/20240523_094951/test/ai-dev_l4t-r35.3.1_test.py.txt; exit ${PIPESTATUS[0]}

sourcing   /opt/ros/humble/install/setup.bash
ROS_DISTRO humble
ROS_ROOT   /opt/ros/humble
testing OpenCV...
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (2.2.1) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (2.2.1) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 95, in <module>
    from urllib3.contrib import pyopenssl
  File "/usr/local/lib/python3.8/dist-packages/urllib3/contrib/pyopenssl.py", line 43, in <module>
    import OpenSSL.SSL  # type: ignore[import-untyped]
  File "/usr/lib/python3/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1553, in <module>
    class X509StoreFlags(object):
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1571, in X509StoreFlags
    NOTIFY_POLICY = _lib.X509_V_FLAG_NOTIFY_POLICY
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_NOTIFY_POLICY'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 72, in apport_excepthook
    from apport.fileutils import likely_packaged, get_recent_crashes
  File "/usr/lib/python3/dist-packages/apport/__init__.py", line 5, in <module>
    from apport.report import Report
  File "/usr/lib/python3/dist-packages/apport/report.py", line 32, in <module>
    import apport.fileutils
  File "/usr/lib/python3/dist-packages/apport/fileutils.py", line 12, in <module>
    import os, glob, subprocess, os.path, time, pwd, sys, requests_unixsocket
  File "/usr/lib/python3/dist-packages/requests_unixsocket/__init__.py", line 1, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 95, in <module>
    from urllib3.contrib import pyopenssl
  File "/usr/local/lib/python3.8/dist-packages/urllib3/contrib/pyopenssl.py", line 43, in <module>
    import OpenSSL.SSL  # type: ignore[import-untyped]
  File "/usr/lib/python3/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1553, in <module>
    class X509StoreFlags(object):
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1571, in X509StoreFlags
    NOTIFY_POLICY = _lib.X509_V_FLAG_NOTIFY_POLICY
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_NOTIFY_POLICY'

Original exception was:
Traceback (most recent call last):
  File "test.py", line 6, in <module>
    import requests
  File "/usr/lib/python3/dist-packages/requests/__init__.py", line 95, in <module>
    from urllib3.contrib import pyopenssl
  File "/usr/local/lib/python3.8/dist-packages/urllib3/contrib/pyopenssl.py", line 43, in <module>
    import OpenSSL.SSL  # type: ignore[import-untyped]
  File "/usr/lib/python3/dist-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1553, in <module>
    class X509StoreFlags(object):
  File "/usr/lib/python3/dist-packages/OpenSSL/crypto.py", line 1571, in X509StoreFlags
    NOTIFY_POLICY = _lib.X509_V_FLAG_NOTIFY_POLICY
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_NOTIFY_POLICY'
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/orin_nx/Desktop/software/jetson-containers/jetson_containers/build.py", line 103, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
  File "/home/orin_nx/Desktop/software/jetson-containers/jetson_containers/container.py", line 162, in build_container
    test_container(name, package, simulate)
  File "/home/orin_nx/Desktop/software/jetson-containers/jetson_containers/container.py", line 322, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --runtime=nvidia --network=host --volume /home/orin_nx/Desktop/software/jetson-containers/packages/opencv:/test --volume /home/orin_nx/Desktop/software/3.1 /bin/bash -c 'python3 test.py' 2>&1 | tee /home/orin_nx/Desktop/software/jetson-containers/logs/20240523_094951/test/ai-dev_l4t-r35.3.1_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```
Full log:
[output_log_jetson_containers .txt](https://github.com/dusty-nv/jetson-containers/files/15420153/output_log_jetson_containers.txt)

I'm unsure what the problem is it looks like some test script failed maybe?

dusty-nv, 2024-05-23:
@mswanson24 I recently had to add this patch to torchvision on 20.04 for the same error, pinning `pyopenssl==24.0.0` (presumably due to some package update on PyPi) - https://github.com/dusty-nv/jetson-containers/commit/004d74c2e3670f6e0a23671ab231eceb08ab5d2b

Can you try adding this same block to the bottom of [`jetson-containers/packages/opencv/install.sh` ](https://github.com/dusty-nv/jetson-containers/blob/master/packages/opencv/install.sh) and try building again?

```
if [ $(lsb_release --codename --short) = "focal" ]; then
    # https://github.com/conda/conda/issues/13619
    pip3 install --no-cache-dir pyopenssl==24.0.0
fi
```

mswanson24, 2024-05-23:
> @mswanson24 I recently had to add this patch to torchvision on 20.04 for the same error, pinning `pyopenssl==24.0.0` (presumably due to some package update on PyPi) - [004d74c](https://github.com/dusty-nv/jetson-containers/commit/004d74c2e3670f6e0a23671ab231eceb08ab5d2b)
> 
> Can you try adding this same block to the bottom of [`jetson-containers/packages/opencv/install.sh` ](https://github.com/dusty-nv/jetson-containers/blob/master/packages/opencv/install.sh) and try building again?
> 
> ```
> if [ $(lsb_release --codename --short) = "focal" ]; then
>     # https://github.com/conda/conda/issues/13619
>     pip3 install --no-cache-dir pyopenssl==24.0.0
> fi
> ```

I will try this and let you know how it goes. Thank you!

mswanson24, 2024-05-28:
Hi @dusty-nv,
I was able to successfully build the image:
```
orin_nx@ubuntu:/media$ docker image ls
REPOSITORY                   TAG                              IMAGE ID       CREATED         SIZE
ai-dev                       l4t-r35.3.1                      c3534d6c6b42   3 days ago      13.5GB
ai-dev                       l4t-r35.3.1-ros_humble-desktop   c3534d6c6b42   3 days ago      13.5GB
ai-dev                       l4t-r35.3.1-cmake                03b5506805e1   3 days ago      11.2GB
ai-dev                       l4t-r35.3.1-opencv_deb           88089638a4de   3 days ago      11.1GB
ai-dev                       l4t-r35.3.1-numpy                753134c78196   3 days ago      10.3GB
```
One thing i noticed in relation to the issue :https://github.com/dusty-nv/jetson-containers/issues/530#issue-2302275837

Is that I am building with r35.3.1 and they are building with r35.4.1.
Could that be causing complications?

closed 2024-06-18 by mswanson24